HKUST-Aerial-Robotics/VINS-Mono #1: Running on odroid XU4

Changliu52, 2017-05-22:
It is really exciting to see this code released! Amazing!
I am working with Odroid XU4. 
I am wondering if it is possible to run it in ARM CPUs directly?
or any changes to be made for that?

Kind Regards,
Chang

qintony, 2017-05-22:
We haven't tried it on Odroid. 
Maybe not because of the computation resources. Also, Ceres may not support arm based architecture.
We have used VINS_mono on TK1, TX1, TX2 successfully.

qintony, 2017-05-22:
I think you can directly clone the code on Odroid and try it under the ROS environment.
No more special dependency.
Use fewer features, may be work.

Changliu52, 2017-05-22:
Awesome. I will definitely try that!
Thank you for your reply.
Chang

Changliu52, 2017-05-22:
Nvidia TK TX is arm based. So should work. If you are not using GPU, they should have similar speed with odroid xu4.

qintony, 2017-05-23:
No GPU.
I suggestion you turn off the loop closure function and limit feature number to 100 at beginning.

longbowlee, 2017-05-23:
This is awesome, is this algorithm support rolling
Shutter cam？

shaojie, 2017-05-23:
Not now, but we will add rolling camera support soon. Stay tuned. 

However, the code runs reasonably well with a rolling shutter camera, as shown in VINS-Mobile (iPhone camera is rolling shutter).

qintony, 2017-05-23:
It depends on how serious the rolling shutter effect is.
You see, our algorithm can directly run with iPhone's cam, which is rolling shutter.

longbowlee, 2017-05-23:
Thanks, Shaojie.
I read part of your PHD dissertation, even I can't fully understand yet. Is this algorithm implemented as indicated in the dissertation, by non-linear optimization rather than filter-based?

maesfahani, 2017-05-23:
Hi,
I am trying to test it on TX1, but the tracking fails and says: "Number of consecutive invalid steps more than Solver::Options::max_num_consecutive_invalid_steps"
I am using MH_01_easy.bag for test and I turned off the loop closure function and limited feature number to 100. However, it runs well on my laptop.

qintony, 2017-05-23:
@maesfahani I have tested again VINS_mono on TX1 ( remove -march=native in CmakeList.txt). I didn't meet this problem. Maybe you environment problem. 
I can only share you with my environment.
3.10.96-tegra Eigen 3.2.9 Ceres 1.11.0  OpenCV 2.4.13
[image]
[image]
Actually, we put TX1 on quadrotor to do onboard experiment.

qintony, 2017-05-23:
@Changliu52 I have tested VINS_Mono on Odroid XU4 (remove -march=native in CmakeList.txt and close loop clusure function). The code can run on Odroid successfully. The performance is not good because of  the limited image processing ability. Some time delay occurs.
I didn't suggest you use Odroid to run vision-based algorithms. 
For TX1, it can achieve real-time performance. What's more, if you know how to use Visionworks, the powerful visual tools on Tegra, you can run faster.  
[image]
[image]

maesfahani, 2017-05-24:
@qintony Thanks.

maesfahani, 2017-05-24:
@qintony is your ROS Kinectic or indigo on TX1?
The ROS on my laptop is Indigo and it works fine.

qintony, 2017-05-24:
@maesfahani  indigo on TX1.

maesfahani, 2017-05-24:
@qintony maybe that is the problem. 
I have ubuntu 16.04 and kinetic on TX1

maesfahani, 2017-05-24:
@qintony did you test it on kinetic before?

qintony, 2017-05-24:
@maesfahani  No. Maybe you need to figure it out by yourself. Thanks.

Changliu52, 2017-05-26:
Interesting. Thank you so much Qin.
You are right. Just found out a57 in tx1 is 15% faster than a15 in odroid xu4... https://www.arm.com/products/processors/cortex-a/cortex-a57-processor.php

Chang

closed 2017-05-30 by Changliu52

fejj, 2017-06-13:
@qintony Confirming I could only run VINS-Mono on the XU4 by disabling loop closure. Image frames were then being processed at 8 fps on average on my data. When you said earlier "performance is not so good", can you confirm you had similar frame rate? Thanks much.

qintony, 2017-06-14:
@fejj  Yes, the image processing rate is lower than 10Hz. We need at least 10 Hz.

MerlotQ, 2017-06-21:
@qintony 
Hi Tong,
I just had test on TK1 however I got the process died error. It seems an assertion failed which shows

`vins_estimator: /usr/local/include/boost/dynamic_bitset/dynamic_bitset.hpp:818: boost::dynamic_bitset<Block, Allocator>& boost::dynamic_bitset<Block, Allocator>::operator^=(const boost::dynamic_bitset<Block, Allocator>&) [with Block = long unsigned int; Allocator = std::allocator<long unsigned int>]: Assertion `size() == rhs.size()' failed.`

The gdb bt info as follows:

#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:44
#1  0xb6476ebe in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#2  0xb6479716 in __GI_abort () at abort.c:89
#3  0xb6472104 in __assert_fail_base (fmt=0x1 <error: Cannot access memory at address 0x1>, 
    assertion=0xee9e0 "size() == rhs.size()", assertion@entry=0x0, 
    file=0xb19c4 "/usr/local/include/boost/dynamic_bitset/dynamic_bitset.hpp", file@entry=0xaf9ff3e0 "\001", 
    line=818, line@entry=3058917540, 
    function=function@entry=0xee734 <boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> >::operator^=(boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> > const&)::__PRETTY_FUNCTION__> "boost::dynamic_bitset<Block, Allocator>& boost::dynamic_bitset<Block, Allocator>::operator^=(const boost::dynamic_bitset<Block, Allocator>&) [with Block = long unsigned int; Allocator = std::allocator"...) at assert.c:92
#4  0xb647219a in __GI___assert_fail (assertion=0x0, file=0xaf9ff3e0 "\001", line=3058917540, 
    function=0xee734 <boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> >::operator^=(boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> > const&)::__PRETTY_FUNCTION__> "boost::dynamic_bitset<Block, Allocator>& boost::dynamic_bitset<Block, Allocator>::operator^=(const boost::dynamic_bitset<Block, Allocator>&) [with Block = long unsigned int; Allocator = std::allocator"...) at assert.c:101
#5  0x000979d8 in operator^= (rhs=..., this=0xaf9fe910)
    at /usr/local/include/boost/dynamic_bitset/dynamic_bitset.hpp:818
#6  operator^<long unsigned int, std::allocator<long unsigned int> > (y=..., x=...)
    at /usr/local/include/boost/dynamic_bitset/dynamic_bitset.hpp:1794
#7  distance (b=..., a=...)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/ThirdParty/DBoW/../DVision/BRIEF.h:164
#8  DBoW2::FBrief::distance (a=..., b=...)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/ThirdParty/DBoW/FBrief.cpp:56
#9  0x0009d41a in DBoW2::TemplatedVocabulary<boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> >, DBoW2::FBrief>::transform (this=0xa556a648, feature=..., word_id=@0xaf9fe9b8: 2898747680, 
    weight=@0xaf9fe9c0: -5.2856594906783572e-47, nid=0xaf9fe9bc, levelsup=2)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/ThirdParty/DBoW/TemplatedVocabulary.h:1237
#10 0x0009e260 in DBoW2::TemplatedVocabulary<boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> >, DBoW2::FBrief>::transform (this=0xa556a648, features=std::vector of length 1938, capacity 1938 = {...}, v=..., fv=..., 
    levelsup=2)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/ThirdParty/DBoW/TemplatedVocabulary.h:1154
#11 0x000a276a in DLoopDetector::TemplatedLoopDetector<boost::dynamic_bitset<unsigned long, std::allocator<unsigned long> >, DBoW2::FBrief>::detectLoop (this=this@entry=0xb0300908, 
    keys=std::vector of length 1938, capacity 2048 = {...}, 
---Type <return> to continue, or q <return> to quit---
    descriptors=std::vector of length 1938, capacity 1938 = {...}, match=..., 
    cur_pts=std::vector of length 0, capacity 0, old_pts=std::vector of length 0, capacity 0)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/TemplatedLoopDetector.h:702
#12 0x0009c108 in run (name="BRIEF", old_index=@0xaf9feafc: -1, old_pts=std::vector of length 0, capacity 0, 
    cur_pts=std::vector of length 0, capacity 0, descriptors=std::vector of length 1938, capacity 1938 = {...}, 
    keys=std::vector of length 1938, capacity 2048 = {...}, this=0xb0300870)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/demoDetector.h:169
#13 LoopClosure::startLoopClosure (this=0xb0300870, keys=std::vector of length 1938, capacity 2048 = {...}, 
    descriptors=std::vector of length 1938, capacity 1938 = {...}, cur_pts=std::vector of length 0, capacity 0, 
    old_pts=std::vector of length 0, capacity 0, old_index=@0xaf9feafc: -1)
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/loop-closure/loop_closure.cpp:31
#14 0x00040d20 in process_loop_detection ()
    at /home/ubuntu/catkin_ws/src/VINS-Mono/vins_estimator/src/estimator_node.cpp:258
#15 0xb6647854 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
 
What might be the best option to solve this? Thanks.

tdnet12434, 2017-07-19:
@fejj 
Did you use emmc for OS ?
I think if running on emmc should get better fps.. maybe.

maciejmatuszak, 2017-07-24:
Hi,
First of all this is great framework! Well written and when works it is really good.
I am trying to run the vins-mono on TX2 module but without success - position drift occurs on the TX2 and TX1.
@qintony since you indicated you run it on TX2 can you please help with solving this. 
I encounter this problem first on TX1 with Jetson 3.0 I assume the reason for fail is same as on TX2. I run with maximum performance settings using jetson_clocks.sh. 

My TX2 env:
- Jetpack 3.0
- Ununtu 16.04
- ros kinetic
from cmake:
-- Found OpenCV: /opt/ros/kinetic (found version "3.2.0")
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found Ceres version: 1.13.0 installed in: /usr/local with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, CXSparse, SchurSpecializations, OpenMP]

I have stereo camera with synchronised IMU (loitor sensor) to narrow the problem I recorded bag file on the TX2. I can run the bag file on laptop without problem, with identical settings / library versions.
. I notice there are debug statement in code so I run it in debug level. With all the debug processing it will fail on the laptop as well but I can see major difference in ceres solver performance.
On TX2 I can get 2-3 iterations but on laptop it will reach the limit of 9 all the time. below is "representative" fragments of logs:

TX2:
```
[32m[DEBUG] [1500876235.516887783, 1500597636.677634330]: new image coming ------------------------------------------[0m
[32m[DEBUG] [1500876235.516928903, 1500597636.677634330]: Adding feature points 64[0m
[32m[DEBUG] [1500876235.516954310, 1500597636.677634330]: input feature: 64[0m
[32m[DEBUG] [1500876235.517000582, 1500597636.677634330]: num of feature: 83[0m
[32m[DEBUG] [1500876235.517095204, 1500597636.677634330]: parallax_sum: 3.985230, parallax_num: 41[0m
[32m[DEBUG] [1500876235.517125348, 1500597636.677634330]: current parallax: 44.712336[0m
[32m[DEBUG] [1500876235.517151460, 1500597636.677634330]: this frame is--------------------accept[0m
[32m[DEBUG] [1500876235.517174403, 1500597636.677634330]: Keyframe[0m
[32m[DEBUG] [1500876235.517196643, 1500597636.677634330]: Solving 10[0m
[32m[DEBUG] [1500876235.517224771, 1500597636.677634330]: number of feature: 83[0m
[32m[DEBUG] [1500876235.517370273, 1500597636.677634330]: triangulation costs 0.068159[0m
[32m[DEBUG] [1500876235.517436384, 1500597636.677634330]: fix extinsic param[0m
[32m[DEBUG] [1500876235.518905163, 1500597636.677634330]: visual measurement count: 546[0m
[32m[DEBUG] [1500876235.518944170, 1500597636.677634330]: prepare for ceres: 1.477803[0m
[32m[DEBUG] [1500876235.556454550, 1500597636.717970814]: Iterations : 2[0m
[32m[DEBUG] [1500876235.556510774, 1500597636.717970814]: solver costs: 37.534092[0m
[32m[DEBUG] [1500876235.561520718, 1500597636.728032654]: pre marginalization 3.708139 ms[0m
[32m[DEBUG] [1500876235.585113696, 1500597636.748159986]: marginalization 23.511475 ms[0m
[32m[DEBUG] [1500876235.586332815, 1500597636.748159986]: whole marginalization costs: 29.564732[0m
[32m[DEBUG] [1500876235.586373390, 1500597636.748159986]: whole time for ceres: 68.906639[0m
[32m[DEBUG] [1500876235.587210114, 1500597636.748159986]: solver costs: 69.883425ms[0m
[32m[DEBUG] [1500876235.587434431, 1500597636.748159986]: marginalization costs: 0.171421ms[0m
[0m[ INFO] [1500876235.587517438, 1500597636.748159986]: position: 0.000440664  0.00050723  -0.0278666[0m
[32m[DEBUG] [1500876235.587566077, 1500597636.748159986]: orientation: -0.00137258    -0.02498   -0.048206[0m
[32m[DEBUG] [1500876235.587647196, 1500597636.748159986]: extirnsic tic:  -0.0287773   0.0556961 0.000498446[0m
[32m[DEBUG] [1500876235.587699675, 1500597636.748159986]: extrinsic ric: -89.5537 -1.72662   179.55[0m
[32m[DEBUG] [1500876235.587728571, 1500597636.748159986]: vo solver costs: 70.691925 ms[0m
[32m[DEBUG] [1500876235.587749659, 1500597636.748159986]: average of time 86.759073 ms[0m
[32m[DEBUG] [1500876235.587769114, 1500597636.748159986]: sum of path 0.039944[0m
[32m[DEBUG] [1500876235.639958934, 1500597636.798498503]: processing vision data with stamp 1500597636.703376 

```
Laptop:
```
[32m[DEBUG] [1500876089.229592989, 1500597636.208239138]: new image coming ------------------------------------------[0m
[32m[DEBUG] [1500876089.229608588, 1500597636.208239138]: Adding feature points 138[0m
[32m[DEBUG] [1500876089.229620616, 1500597636.208239138]: input feature: 138[0m
[32m[DEBUG] [1500876089.229631783, 1500597636.208239138]: num of feature: 155[0m
[32m[DEBUG] [1500876089.229675986, 1500597636.208239138]: parallax_sum: 1.656623, parallax_num: 118[0m
[32m[DEBUG] [1500876089.229684131, 1500597636.208239138]: current parallax: 6.458021[0m
[32m[DEBUG] [1500876089.229691790, 1500597636.208239138]: this frame is--------------------reject[0m
[32m[DEBUG] [1500876089.229698299, 1500597636.208239138]: Non-keyframe[0m
[32m[DEBUG] [1500876089.229704149, 1500597636.208239138]: Solving 10[0m
[32m[DEBUG] [1500876089.229711940, 1500597636.208239138]: number of feature: 155[0m
[32m[DEBUG] [1500876089.229754109, 1500597636.208239138]: triangulation costs 0.001732[0m
[32m[DEBUG] [1500876089.229773620, 1500597636.208239138]: fix extinsic param[0m
[32m[DEBUG] [1500876089.230524258, 1500597636.208239138]: loop constraint num: 0[0m
[... cut ...]
[32m[DEBUG] [1500876089.254743458, 1500597636.228371302]: sum of path 0.011036[0m
[32m[DEBUG] [1500876089.304051829, 1500597636.278650193]: processing vision data with stamp 1500597636.206887 
```

For the above logs on TX2 I limited the max_cnt: 100 and disabled the loop closure loop_closure: 0, on  laptop those are default 150 and loop closure enabled.

I tried to play with lowrering the "freq"  and extending the "max_solver_time" but did not got good results.
I notice the parallel execution of ceres is disabled could that help?

Maciej