HKUST-Aerial-Robotics/VINS-Mono #4: Can't run vins_rviz.launch successfully

wangqi123412431, 2017-05-24:
Hi, I'm very glad to see this open-source project and can't wait to run it on my PC. Thanks for your contribution to the community!

But when I run " roslaunch vins_estimator vins_rviz.launch" after another two  step("rosbag play xxx.bag" and "roslaunch vins_estimator euroc.launch" ) which have been run successfully, I met an error:

[image]

The node run in the vins_rviz.launch is  "rvizvisualisation" which belongs to package rviz, but I can't search the node in my indigo ROS. I replace "rvizvisualisation" with node "rviz", but the same error occured. If I just run "rosrun rviz rviz" by not adding "vins_rviz_config.rviz" , it works. Don't the config file and node "rviz" match each other? 
How can I solve this problem? Where can I get the rvizvisualisation node? 

Thank you!!!

qintony, 2017-05-24:
1. check the path /home/wangqi/catkin_ws/src/VINS-Mono-master/vins_estimator/../config/vins_rviz_config.rviz is right or not...
2. This launch file is not necessary. You can directly open RVIZ, add the corresponding topics.

[image]

wangqi123412431, 2017-05-24:
Oh...the rvizvisualisation is just a name!! All right, forgive me, it was careless of me...

Fortunately, I solved the problem by deleting "-d" in vins_rviz.launch(My file path is right and the rviz is newest version)，now it is:
`<launch>`
    `<node name="rvizvisualisation" pkg="rviz" type="rviz" output="log" args="$(find vins_estimator)/../config/vins_rviz_config.rviz" />`
`</launch>`

Thanks for your answer.

qintony, 2017-05-24:
@wangqi123412431 
It should be "-d". see the help menu of RVIZ. Anyway, it doesn't matter....

tony-ws@tony-ws:~$ rosrun rviz rviz --help
[ INFO] [1495607652.611493847]: rviz version 1.11.15
[ INFO] [1495607652.611530033]: compiled against Qt version 4.8.6
[ INFO] [1495607652.611547018]: compiled against OGRE version 1.8.1 (Byatis)
rviz command line options:
  -h [ --help ]               Produce this help message
  -s [ --splash-screen ] arg  A custom splash-screen image to display
  --help-file arg             A custom html file to show as the help screen
  -d [ --display-config ] arg A display config file (.rviz) to load

liyanjie-beijing, 2017-05-30:
I met the same problem:
[image]

qintony, 2017-05-30:
@liyanjie-beijing 
try deleting "-d" in vins_rviz.launch
or open RVIZ and add topics manually.
RVIZ is just for visualization.

wangqi123412431, 2017-06-01:
oh...sorry, I find deleting "-d" isn't a good way to solve this problem, although I can open the rviz without error by deleting "-d", the interface of rviz is not same with your original one which includes many necessary topics, what I opened is a basic rviz which only includes one topic -- vins_estimator/path...this means I didn't open the config file successfully.

I don't know the reason so far, but like what you said, RVIZ is just for visualisation, I added these topics manually, and now I focus on the algorithms ~

MAGUADIDI, 2018-01-14:
[image]
[image]
@wangqi123412431 @liyanjie-beijing @qintonguav  can I sovle it?thanks.

qintonguav, 2018-01-14:
@MAGUADIDI 

source ~/catkin_ws/devel/setup.bash

MAGUADIDI, 2018-01-14:
Thank you..But it never work..
[image]

@qintonguav

sheroy666, 2018-04-21:
@qintonguav 
I can have this . But when I rosbag play MH_05_difficult.bag ,there is no tracked image.Could u please tell me why?
[image]

brandnew2017, 2019-02-27:
So I want to ask if you find good way to solve this problem???

thank you!!
[image]

RyanMiao11, 2019-08-30:
> @MAGUADIDI
> 
> source ~/catkin_ws/devel/setup.bash

I do this.but it still exists！