HKUST-Aerial-Robotics/VINS-Mono #5: Is it possible to use it on an AR Drone 2.0

JiaoYanMoGu, 2017-05-24:
Is it possible to use it on AR Drone UAV by using  the IMU and Camera itself ? How to make the IMU synchronized ? I'm really a noob. Thanks!

qintony, 2017-05-24:
AR Drone itself don't have the computation ability to run VINS.

You can consider ground station. Collect the image and IMU by AR Drone, and send the data to ground PC. The transmission may be a problem.

Synchronization is designed by hardware factory, you can do nothing.

The next generation of DJI controller may have the IMU-camera synchronization ability.

JiaoYanMoGu, 2017-05-24:
Sorry, I just said is wrong，What I mean is that run on a PC too。 And what kind of IMU-camera do you use ? Thanks

qintony, 2017-05-24:
You can see our device in the video link. 

Also, you can find in 
https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/config/extrinsic_parameter_example.pdf
The second one in pdf.

lrbdh, 2017-05-30:
Hi, can I use other USB cameras?

qintony, 2017-05-30:
yes, you can. 
Try to use global shutter camera with accurate time stamp for better performance.

lrbdh, 2017-05-30:
OK, Thanks!

amc92, 2017-11-30:
Hi,
I am also trying to run RVIZ in my computer with the data received from an ARDrone. Did you ever get it working properly? Thanks!!