HKUST-Aerial-Robotics/VINS-Mono #6: code problem

zdczdcc, 2017-05-24:
In the function `FeatureTracker::readImage(const cv::Mat &_img)`,the variable `img_cnt` doesn't make any change all the time.So the `prev_img` is equals to `curr_img`.I wonder if this is a bug.

tanmengwen, 2017-05-24:
Yes, img_cnt doesn't make any change all the time.
But the prev_img will be changed by the forw_img at the end of the function of FeatureTracker::readImage.

However, I have a question how does the feature_tracker node get the init cur_pts, because I didn't find the feature detector in the feature_tracker. In the function BriefExtractor::operator(), there has a FAST corner detector, but it will be disable when I close the loop closure!

Maybe I miss something, I hope someone can answer my question. @qintony

PeiliangLi, 2017-05-24:
hi @zdczdcc, it's not a bug, img_cnt was used to control the frequency of the feature detector in the previous version. And @tanmengwen goodfeaturetotrack() is for feature detection.

zdczdcc, 2017-05-24:
Hi @PeiliangLi ，
I've found that in the ios version of VINS,the public member variable `img_cnt` is changed in file `ViewController.mm`.

As your said,the `img_cnt` is always`=0`.In that case,the `prev_img` is same as `cur_img` all the time.In my opinion,there is no longer to have 2 different meaning variable to save one same variable even they both are needed.It would make it difficult to understand.

zdczdcc, 2017-05-24:
@tanmengwen you could read their papers for feature detection.

PeiliangLi, 2017-05-24:
@zdczdcc yes, it may lead a little confusion, we will modify here later.

qintony, 2017-05-24:
@zdczdcc Thanks for your issue. @tanmengwen Thanks for your answer.

I rewrite this part code, you can see my last commit. 

I remove the img_cnt, replace it with a bool flag, PUB_THIS_FRAME. The meaning will be clear.

zdczdcc, 2017-05-25:
Hi @qintony ,
i've read more with your latest commit.Now the feature_tracker wouldn't track new corners while the image frequency is larger than imu frequency.Thanks for your modification.

closed 2017-05-25 by zdczdcc