HKUST-Aerial-Robotics/VINS-Mono #7: catkin_make error

cumtchenchang, 2017-05-24:
ubuntu 16.04 kinetic 
when i use catkin_make, i encountored error as follows:
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetSearchPosition
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetDistanceToObstacle
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetRecoveryInfo
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetRobotTrajectory
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _hector_nav_msgs_generate_messages_check_deps_GetNormal
Scanning dependencies of target benchmark_publisher
Scanning dependencies of target camera_model
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_cpp
Scanning dependencies of target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target _catkin_empty_exported_target
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_eus
Scanning dependencies of target rosgraph_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_nodejs
Scanning dependencies of target roscpp_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_cpp
Scanning dependencies of target roscpp_generate_messages_lisp
Scanning dependencies of target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target roscpp_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
Scanning dependencies of target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_nodejs
[  5%] Built target rplidarNode
[  6%] Built target rplidarNodeClient
[  7%] Built target imu_attitude_to_tf_node
[  9%] Built target map_to_image_node
[  9%] Built target visualization_msgs_generate_messages_cpp
Scanning dependencies of target tf2_msgs_generate_messages_py
[  9%] Built target tf2_msgs_generate_messages_py
Scanning dependencies of target tf2_msgs_generate_messages_eus
[  9%] Built target tf2_msgs_generate_messages_eus
Scanning dependencies of target actionlib_generate_messages_py
[  9%] Built target actionlib_generate_messages_py
Scanning dependencies of target actionlib_generate_messages_nodejs
[  9%] Built target actionlib_generate_messages_nodejs
Scanning dependencies of target actionlib_generate_messages_lisp
[  9%] Built target actionlib_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_cpp
[  9%] Built target actionlib_generate_messages_cpp
[ 11%] Built target pose_and_orientation_to_imu_node
Scanning dependencies of target sensor_msgs_generate_messages_py
Scanning dependencies of target tf2_msgs_generate_messages_cpp
[... 74 lines cut ...]
[ 72%] Linking CXX static library /home/cc/catkin_ws/devel/lib/libcamera_model.a
[ 72%] Built target camera_model
Scanning dependencies of target vins_estimator
Scanning dependencies of target ar_demo_node
Scanning dependencies of target feature_tracker
[ 72%] Building CXX object VINS-Mono/ar_demo/CMakeFiles/ar_demo_node.dir/src/ar_demo_node.cpp.o
[ 73%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/feature_tracker_node.cpp.o
[ 73%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/parameters.cpp.o
[ 74%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o
/home/cc/catkin_ws/src/VINS-Mono/vins_estimator/src/estimator_node.cpp:18:38: fatal error: diagnostic_msgs/KeyValue.h: No such file or directory
compilation terminated.
VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/build.make:62: recipe for target 'VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o' failed
make[2]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o] Error 1
CMakeFiles/Makefile2:4361: recipe for target 'VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all' failed
make[1]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 75%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/feature_tracker.cpp.o
[ 76%] Linking CXX executable /home/cc/catkin_ws/devel/lib/feature_tracker/feature_tracker
[ 77%] Linking CXX executable /home/cc/catkin_ws/devel/lib/ar_demo/ar_demo_node
[ 77%] Built target ar_demo_node
[ 77%] Built target feature_tracker
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
cc@cc-K55VD:~/catkin_ws$

qintony, 2017-05-24:
Thanks for your issue.
Two solutions.
1. sudo apt-get install ros-kinetic-diagnostic-msgs
    sudo apt-get install ros-kinetic-diagnostics
2. just remove #include <diagnostic_msgs/KeyValue.h> in estimator_node.cpp:18

FangGet, 2017-05-24:
sudo apt-get install ros-kinetic-diagnostic-msgs

it just solved the problem~

closed 2017-05-24 by qintony