HKUST-Aerial-Robotics/VINS-Mono #8: Can't build  successfully

MC215, 2017-05-30:
I already install the dependency for eigen, ceres, but when I catkin_make, error

[  4%] Built target benchmark_publisher
[  6%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/chessboard/Chessboard.cc.o
[  8%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/calib/CameraCalibration.cc.o
[ 10%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/Camera.cc.o
[ 12%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CameraFactory.cc.o
[ 16%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/EquidistantCamera.cc.o
[ 16%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CostFunctionFactory.cc.o
[ 18%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/PinholeCamera.cc.o
[ 20%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CataCamera.cc.o
[ 22%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/ScaramuzzaCamera.cc.o
[ 25%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/sparse_graph/Transform.cc.o
[ 27%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/gpl.cc.o
[ 29%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/EigenQuaternionParameterization.cc.o
[ 31%] Linking CXX shared library /home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so
[ 31%] Built target camera_model
[ 35%] Linking CXX executable /home/mc/SLAM/catkin_ws/devel/lib/ar_demo/ar_demo_node
[ 35%] Linking CXX executable /home/mc/SLAM/catkin_ws/devel/lib/feature_tracker/feature_tracker
Scanning dependencies of target vins_estimator
[ 39%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator.cpp.o
[ 39%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o
[ 43%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/feature_manager.cpp.o
[ 43%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/parameters.cpp.o
[ 45%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/factor/pose_local_parameterization.cpp.o
[ 47%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/factor/projection_factor.cpp.o
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `vtable for ceres::CauchyLoss'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solver::Summary::Summary()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::AddResidualBlock(ceres::CostFunction*, ceres::LossFunction*, double*, double*, double*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::SetParameterization(double*, ceres::LocalParameterization*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::~Problem()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solver::Summary::FullReport() const'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::Problem()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessage::stream()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `typeinfo for ceres::LocalParameterization'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int)'
/home/mc/SLAM/catkin_ws/devel/lib/collect2: error: ld returned 1 exit status
libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `vtable for ceres::CauchyLoss'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::LocalParameterization::~LocalParameterization()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solver::Summary::Summary()'
/home/mc/SLAM/catkin_ws/devel/make[2]: lib/libcamera_model.so:*** [/home/mc/SLAM/catkin_ws/devel/lib/ar_demo/ar_demo_node] Error 1 
undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::AddResidualBlock(ceres::CostFunction*,make[1]:  *** [VINS-Mono/ar_demo/CMakeFiles/ar_demo_node.dir/all] Error 2
ceres::LossFunction*, double*, make[1]: double*, double*)'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::SetParameterization(double*, ceres::LocalParameterization*)'
/home/mc/*** Waiting for unfinished jobs....
SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::~Problem()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Solver::Summary::FullReport() const'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::Problem::Problem()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessage::stream()'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `ceres::LocalParameterization::MultiplyByJacobian(double const*, int, double const*, double*) const'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `typeinfo for ceres::LocalParameterization'
/home/mc/SLAM/catkin_ws/devel/lib/libcamera_model.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/mc/SLAM/catkin_ws/devel/lib/feature_tracker/feature_tracker] Error 1
make[1]: *** [VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/all] Error 2

how can I should do

qintony, 2017-05-30:
check your ceres version.
make install when you install ceres.

ZhangXinNan, 2019-04-19:
@MC215 which version of ceres is right ?