HKUST-Aerial-Robotics/VINS-Mono #9: catkin_make failed

supersheepcy, 2017-05-31:
Hi,I've already install ceres and other dependencies,however I cannot catkin_make sucessfully. How can I solve these problems? Thank you.

Base path: /home/ubuntu/vins-ws
Source space: /home/ubuntu/vins-ws/src
Build space: /home/ubuntu/vins-ws/build
Devel space: /home/ubuntu/vins-ws/devel
Install space: /home/ubuntu/vins-ws/install
####
#### Running command: "cmake /home/ubuntu/vins-ws/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/vins-ws/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/vins-ws/install -G Unix Makefiles" in "/home/ubuntu/vins-ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/vins-ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/vins-ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - benchmark_publisher
-- ~~  - camera_model
-- ~~  - ar_demo
-- ~~  - feature_tracker
-- ~~  - vins_estimator
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'benchmark_publisher'
-- ==> add_subdirectory(VINS-Mono/benchmark_publisher)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'camera_model'
-- ==> add_subdirectory(VINS-Mono/camera_model)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   program_options
--   system
-- Found OpenCV: /opt/ros/kinetic (found version "3.2.0") 
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE - Success
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /usr/local with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP]
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'system_lib' but neither
  'system_lib_INCLUDE_DIRS' nor 'system_lib_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  VINS-Mono/camera_model/CMakeLists.txt:23 (catkin_package)

-- +++ processing catkin package: 'ar_demo'
-- ==> add_subdirectory(VINS-Mono/ar_demo)
-- +++ processing catkin package: 'feature_tracker'
-- ==> add_subdirectory(VINS-Mono/feature_tracker)
-- +++ processing catkin package: 'vins_estimator'
-- ==> add_subdirectory(VINS-Mono/vins_estimator)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found required Ceres dependency: Eigen version 3.2.92 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /usr/local with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/vins-ws/build
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/vins-ws/build"
####
Scanning dependencies of target benchmark_publisher
Scanning dependencies of target camera_model
[  2%] Building CXX object VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o
[  4%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/chessboard/Chessboard.cc.o
[  8%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/calib/CameraCalibration.cc.o
[  8%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/Camera.cc.o
[ 10%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CameraFactory.cc.o
/home/ubuntu/vins-ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp: In function ‘int main(int, char**)’:
/home/ubuntu/vins-ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:141:25: warning: ignoring return value of ‘char* fgets(char*, int, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(tmp, 10000, f);
                         ^
/home/ubuntu/vins-ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp: In constructor ‘Data::Data(FILE*)’:
/home/ubuntu/vins-ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:42:30: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
                &ax, &ay, &az);
                              ^
[ 12%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CostFunctionFactory.cc.o
[ 14%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/PinholeCamera.cc.o
[ 16%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CataCamera.cc.o
[ 18%] Linking CXX executable /home/ubuntu/vins-ws/devel/lib/benchmark_publisher/benchmark_publisher
[ 20%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/EquidistantCamera.cc.o
[ 20%] Built target benchmark_publisher
[ 22%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/ScaramuzzaCamera.cc.o
[ 25%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/sparse_graph/Transform.cc.o
[ 27%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/gpl.cc.o
[ 29%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/EigenQuaternionParameterization.cc.o
[ 31%] Linking CXX shared library /home/ubuntu/vins-ws/devel/lib/libcamera_model.so
[ 31%] Built target camera_model
Scanning dependencies of target ar_demo_node
Scanning dependencies of target feature_tracker
Scanning dependencies of target vins_estimator
[ 35%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/feature_tracker_node.cpp.o
[ 35%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/parameters.cpp.o
[ 37%] Building CXX object VINS-Mono/ar_demo/CMakeFiles/ar_demo_node.dir/src/ar_demo_node.cpp.o
[ 39%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o
[ 41%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/parameters.cpp.o
[ 47%] Building CXX object VINS-Mono/feature_tracker/CMakeFiles/feature_tracker.dir/src/feature_tracker.cpp.o
[ 47%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator.cpp.o
[ 47%] Linking CXX executable /home/ubuntu/vins-ws/devel/lib/ar_demo/ar_demo_node
[ 47%] Built target ar_demo_node
[ 50%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/feature_manager.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/build.make:62: recipe for target 'VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o' failed
make[2]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
[ 52%] Linking CXX executable /home/ubuntu/vins-ws/devel/lib/feature_tracker/feature_tracker
[ 52%] Built target feature_tracker
CMakeFiles/Makefile2:2166: recipe for target 'VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all' failed
make[1]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

PeiliangLi, 2017-05-31:
Have you tried running command: "catkin_make " in "/home/ubuntu/vins-ws/" directory?

supersheepcy, 2017-05-31:
Yes,of course. I just run catkin_make in the directory of  "/home/ubuntu/vins-ws/",and I met errors above.

qintony, 2017-05-31:
@supersheepcy  try  "catkin_make -j1"
In what platform you are making vins_mono?
It seems that a small memory or swap space for compile.

qintony, 2017-05-31:
or compile the packages one by one.
camera_model->feature_tracker->vins_estimator->ar_demo.

closed 2017-06-15 by supersheepcy

lieyanzhan, 2023-06-27:
> or compile the packages one by one. camera_model->feature_tracker->vins_estimator->ar_demo.

how to compile the package one by one?

lieyanzhan, 2023-06-27:
I have solve it, please check [this blog](https://blog.csdn.net/bjydlcq/article/details/131425448?csdn_share_tail=%7B%22type%22%3A%22blog%22%2C%22rType%22%3A%22article%22%2C%22rId%22%3A%22131425448%22%2C%22source%22%3A%22bjydlcq%22%7D)