HKUST-Aerial-Robotics/VINS-Mono #10: What does the each item in output file "vins_result.csv" mean in ros？

txy362922388, 2017-06-05:
I know the first three items are position (x,y,z),how about next items ?

PeiliangLi, 2017-06-05:
Please find it in https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/utility/visualization.cpp  pubOdometry()

qintony, 2017-06-05:
You can find the result in the source code which Peiliang tells you.

time stamp + position(x,y, z) + quaternion(qw, qx, qy, qz) + velocity(x,y,z)

all of them are in world frame (where the z-axis is gravity direction)

closed 2017-06-06 by qintony

alexis-dupuis, 2018-06-06:
I just want to update this answer for anyone still interested in it. Vins-Mono now outputs 2 result files whose names are pretty explicites:

- "vins_result_loop" : defined in [path to Vins folder]/pose_graph/src/pose_graph.cpp ; line 156. The format is timestamp + position(x,y, z) + quaternion(qw, qx, qy, qz).

- "vins_result_no_loop": defined in [path to Vins folder]/vins_estimator/src/utility/visualization.cpp in function pubOdometry(). The format is timestamp + position(x,y, z) + quaternion(qw, qx, qy, qz) + velocity(x,y,z).

They are published in the folder defined by the parameter "output_path" in your yaml file, for instance: [path to VINS folder]/config/euroc/euroc_config.yaml ; line 6. 

Hope this may clear things for anyone.