HKUST-Aerial-Robotics/VINS-Mono #11: no results when running with my device

yuoct, 2017-06-05:
Hi, I want to run with my camera and IMU, and I have change the topic name in the config file. However there is no output about position information. And the Rviz shows as:
[image]

I find that the Global Status is abnormal and it shows: No tf data. Acutual error: Fixed Frame [world] doesn't exist. 
So what's the problem and thanks for your help.

qintony, 2017-06-05:
Check the time stamp in your Camera and IMU topic.

The header of these two topics should contain the accurate time stamps.

supersheepcy, 2017-06-15:
@qintony Should the time stamps of these two topics be the same ?

qintony, 2017-06-15:
Yes. Under the same system clock.

shibowing, 2017-06-18:
hi, i am a beginner of ros . Thanks for your advice. I know how to increase the frequency of imu. However, i have puzzled about how to synchronize the imu and camera.
Actually, i have a xtion camera and xsens imu. I just launched the xsens driver and openni driver at the same time. Is that a way of synchronization? Can you tell me a way to synchronize the imu and camera? Thanks for your advice @qintony

shibowing, 2017-06-18:
Hi, i am a beginner of ros .  I have puzzled about how to synchronize the imu and camera message.
Actually, i have a xtion camera and a xsens imu. I just launched the xsens driver and openni driver at the same time. Is that a way of synchronization? Can you tell me a way to synchronize the imu and camera? Thanks for your advice @qintony

caoxudong0513, 2017-07-13:
我也遇到了这个问题，请问你解决这个问题了吗？

closed 2017-07-14 by qintony