HKUST-Aerial-Robotics/VINS-Mono #12: cannot find where tic is calibrated?

highlightz, 2017-06-05:
In your paper _Monocular Visual-Inertial State Estimation With Online Initialization and Camera-IMU Extrinsic Calibration_, you've said that rotation matrix between IMU and Cam is calibrated at first phase, and translation tic is calibrated at second phase. And I've run your code using EuRoC dataset with ESTIMATE_EXTRINSIC set to 2, and the estimator gives an acceptable calibration result for both ric and tic. However, when I read about the code(estimator.cpp), I cannot find the tic's calibration process in anywhere of the code, and can only find process of ric calibraion(shown below).
`if(ESTIMATE_EXTRINSIC == 2)
    {
        ROS_INFO("calibrating extrinsic param, rotation movement is needed");
        if (frame_count != 0)
        {
            vector<pair<Vector3d, Vector3d>> corres = f_manager.getCorresponding(frame_count - 1, frame_count);
            Matrix3d calib_ric;
            // NOTE: only calib rotation in this stage, not including translation.
            // TOASK: where is t_i_c calibrated?
            if (initial_ex_rotation.CalibrationExRotation(corres, pre_integrations[frame_count]->delta_q, calib_ric))
            {
                ROS_WARN("initial extrinsic rotation calib success");
                ROS_WARN_STREAM("initial extrinsic rotation: " << endl << calib_ric);
                ric[0] = calib_ric;
                RIC[0] = calib_ric;
                ESTIMATE_EXTRINSIC = 1;
            }
        }
    }`
So, could you please tell me where I can find the tic calibration part? Thanks for your excellent work.

qintony, 2017-06-05:
The tic's calibration is in nonlinear optimization implicitly.

Focus on "para_Ex_Pose" in "optimization" function.

[image]

highlightz, 2017-06-06:
Got it. thanks a lot.

closed 2017-06-06 by qintony

AurelianApet, 2019-03-11:
> The tic's calibration is in nonlinear optimization implicitly.
> 
> Focus on "para_Ex_Pose" in "optimization" function.
> 
> [image]