HKUST-Aerial-Robotics/VINS-Mono #13: The z axis of camera coordinate does not aline with the x axia of world coordinate

pssp123, 2017-06-06:
The z axis of camera coordinate does not aline with the x axia of world coordinate.
How can I do?

qintony, 2017-06-06:
The world coordinate is fixed, where the z-axis is along with gravity.  
The camera coordinate is arbitrarily moving with the camera.

I don't fully understand your question, why these two axes should be aligned?

pssp123, 2017-06-06:
In my project, the X-axis of the world coordinate is the opposite of the Z-axis of the camera
coordinate.When I move forward the camera,the data of x is the negative.When I move back the camera,
the data of x is the positive.It looks like  uncomfortable...我好想说中文~

qintony, 2017-06-06:
That's because the x-axis of IMU is opposite with z-axis of the camera. 
When you move forward the camera, actually, the IMU itself go backward. We treat IMU center as body frame. So in the world frame. you go back.
To solve this, you can turn your IMU 180 degrees.

pssp123, 2017-06-07:
It`s not that the x-axis of  IMU is opposite with z-axis of the camera.Actually,it`s the z-axis of the IMU opposited of the z-axis of the camera.
The right rotation matrix transfrom camera to IMU is [ 0 -1 0  -1 0 0   0 0 -1]. 
Can I change it to [ 0 -1 0  -1 0 0   0 0 1] in config to solove this problem?
Because the IMU and camera are fixed.

qintony, 2017-06-07:
Absolutely impossible. It's not an SE3, [ 0 -1 0 -1 0 0 0 0 1].

you can try
Right-multiply the outputs by Utility::ypr2R(Eigen::Vector3d(180,0,0))

pssp123, 2017-06-07:
Thanks.But why the rotation matrix [ 0 -1 0 -1 0 0 0 0 1] cant constitue SE3 with the t matrix？How to confirm it？

qintony, 2017-06-07:
det( [ 0 -1 0; -1 0 0; 0 0 1] ) = -1

which is left-hand coordinate.

What we need is det() = 1, the right-hand coordinate.