HKUST-Aerial-Robotics/VINS-Mono #14: Question about mid-point integration in integration_base.h

highlightz, 2017-06-06:
In your paper _TRO VINS-Mono: A Robust and Versatile Monocular Visual-Inertial State Estimator_, you've demonstrated the Euler integration procedure in an understandable and elegant manner. At the same time, your implementation chooses Mid-point integration, as is shown in VINS-Mono project. 
In method IntegrationBase::midPointIntegration(), I see that you propagate jacobian and covariance with similar formular as (10) and (11) in _TRO_, 
[image]
[image]
I mean, 
[image]

What I do not understand is that how F and V are computed? 
[image]

Are there corresponding formulas in your other papers?  I'd appreciate it if you can provide any information about this. Thank much again for your outstanding work.

qintony, 2017-06-06:
[image]
[image]

Second order approximation, derived by myself.

highlightz, 2017-06-07:
@qintony Yeah, your derivation exactly matches your implementation. I am still confused about the calculation of f01, f31, etc. Forgive me for my poor math, however, could you please give me an example of the computation of f01? Thanks a lot.

qintony, 2017-06-07:
You maybe confuse about quaternion derivate.
You can refer to the following papers.

Quaternion kinematics for the error-state KF
http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf

Similiar step in Page.51

highlightz, 2017-06-07:
Thanks for your great help. You're correct, the quaternion derivate does confuse me. I read through the _kienmatics.pdf_ paper, and know the meaning of your formula. Best wishes.