HKUST-Aerial-Robotics/VINS-Mono #15: vins_estimator fail to run

mlyarthur, 2017-06-07:
Hello!when i run the vins_estimator node for the first time ,there is an error of Eigen  ,and i think i solve it following the instruction of http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html.
However it still failed with another error like this .

and the second picture is information with GDB debugger

the error seems to happen when delete
[image]
 the pointer,but i dont modify any code .
[image]

mlyarthur, 2017-06-07:
i am sorry it failed to upload the picture .so i try again to show them.

[image]
the second picture is information with GDB debugger 
[image]

qintony, 2017-06-07:
Sorry, I haven't met your problem.
Maybe the compatibility problem of Eigen or Ceres. 
Try to use the same version of what I use. If the problem still exists, tell me later.

mlyarthur, 2017-06-07:
... but I check the version of  Eigen or Ceres on my computer,which are 3.2.0 and 1.12.0 respectively.And i followed the ceres solver installation instruction strictly .
i solve the previous  eigen problem by adding this macro in IntegrationBase class .
[image]
but i dont know if there is a connection between the `new` problem and this old one 
i use gdb to find the location  where it crash 
delete it_0->second.pre_integration;
[image]

qintony, 2017-06-07:
Actually, you can delete this sentence, "delete it_0->second.pre_integration". It doesn't matter except very little memory.

Try again.

mlyarthur, 2017-06-07:
sorry ,i want to ask a easy question about the version of Eigen. i just use apt-get  libeigen3-dev  
which indicate the version of 3.2.0
[image]

but when i cmake ,it says that the version is 3.3.3 in  following picture
[image]

qintony, 2017-06-07:
In fact, there are two Eigen in your system, which causes the problem.
One is in /usr/include/eigen3/ ... another one is in /usr/local/include/eigen3/
The cmake file will help you to find one Eigen
https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/support_files/cmake/FindEigen.cmake
Eigen 3.3.3 is its finding result.

You can remove Eigen 3.3.3 or hard code the path of Eigen in CMakeLists.  

set(EIGEN_INCLUDE_DIR "/usr/include/eigen3")
include_directories(${CERES_INCLUDE_DIRS})

Make sure Ceres and VINS find the same Eigen 3.2...

mlyarthur, 2017-06-08:
I change the version of eigen to 3.2.0
[image]
but the previous problem happens again so i think  the version doesnt matter
[image]
it is the location of the  problem i meet ,it says eigen have some allign error.i am still confused.
[image]

qintony, 2017-06-08:
can you use 
cat /usr/local/include/eigen3/Eigen/src/Core/util/Macros.h | grep VERSION
to check your Eigen version.

I doubt two versions of Eigen is mixed on your computer.

mlyarthur, 2017-06-08:
this command give  me the result like this 
[image]
it means two versions are both 3.2.0.so....

qintony, 2017-06-08:
rebuild Ceres.

The Ceres which you build previously may depend on 3.3... Now you need to rebuild it, to make sure Ceres and VINS depend on the same Eigen 3.2

mlyarthur, 2017-06-08:
In fact i rebuild Ceres an VINS  when i remove the version 3.3.3 of eigen ,and i now make sure both of them have the same Eigen depencies bacause only one version exsits now .the eigen is located in /usr/include.And there is still error with eigen which keep the same with previous one.

qintony, 2017-06-12:
Try remove devel and build folder in catkin_ws, catkin_make again.

mlyarthur, 2017-06-12:
Do you mean VINS-MONO uses /home/exbot/catkin_ws/src/VINS-Mono/support_files/cmake/FindEigen.cmake to find the location of Eigen?

qintony, 2017-06-12:
I am afraid some people can't find the same Eigen with Ceres in their computer, so I additionally put a FindEigen.cmake file in support files to help them.

mlyarthur, 2017-06-12:
I want to know the system version of Ubuntu,and i work on Ubuntu 14.04 32Bit

mlyarthur, 2017-06-12:
I doubt if there is problem with system version,do you work on 32bit system?

qintony, 2017-06-12:
I haven't tested it on the 32bit system.  What I use is 14.04 and 16.04 with 64 bit.
Maybe is the 32-bit issue.
Thanks.

mlyarthur, 2017-06-14:
I test it on the ubuntu 16.04 64bit successfully...thanks for your kind instruction.

mlyarthur, 2017-06-14:
I test it on the ubuntu 14.04 64bit successfully...thanks for your kind instruction.

closed 2017-06-14 by mlyarthur