HKUST-Aerial-Robotics/VINS-Mono #16: There is big shift when I used my own sensors

YanMinbit, 2017-06-07:
The following picture shows the trajectory generated by VINS-Mono when I moved my camera and IMU rig in the range of 1 square meter. 
[image]

What may be the reason for such a big drift? 
The IMU I used is Mti, the third generation product of Xsens. The camera I used is Point Grey's Color Blackfly USB 3.0 camera.
 I have  tried to set the frequency of IMU to 120 Hz and 256 Hz, the max frequency of the Mti IMU, and the frequency of camera to 5 Hz, 10 Hz, 20 Hz, 30 Hz. When the IMU and the camera were set to 256Hz and 5Hz, the best results will get, but the big drift can not avoid through changing the frequency. 
I guess there may be several reasons: The camera and IMU are unsynchronized;  Camera calibration parameters are not ideal; IMU output data is problematic. What about your opinions?
I observed that you have pointed out that the performance is different on different devices :(global shutter camera + synchronized high-end IMU, e.g. VI-Sensor) > (global shutter camera + synchronized low-end IMU, e.g. camera+DJI A3) > (global camera + unsync high frequency IMU) > (global camera + unsync low frequency IMU) > (rolling camera + unsync low frequency IMU). How big is the difference？ Can you provide some degree of expression?

qintony, 2017-06-07:
From the result, it can be seen that the estimator is unstable, sometimes diverge...
In general, the frequency of camera should be more than 20Hz. At least 20Hz.
Why 5Hz you get the best result, that's because the time offset is trivial compared with large frame interval.
The most possible reason is unsynchronized sensors. 
You can try to give the accurate extrinsic parameters between camera and IMU and accurate intrinsic parameters of the camera. Fix the extrinsic parameters.
Also, if the time offset of camera and IMU is constant, you can manually compensate the time of camera from -10ms to 10ms to find it.

qintony, 2017-06-07:
(global camera + unsync high frequency IMU) 
This sensor set is used In the AR demo, if you have run the AR demo, you will find a good result can be got. The IMU in the AR demo bag is unsynchronized with high frequency.

 (rolling camera + unsync low frequency IMU)
I have tried with DJI M100 (100Hz unsync IMU) and logi-tech camera. Only when the motion is very slowly and smooth, it can work. Very unstable.

Dorothy-2016, 2017-06-12:
Sorry I'm a newer to ROS, I have sucessfully run the data given, now I want to run on my own camera and IMU, How can I modify the camera topic and the IMU topic? Can you give me some advice?@qintony, I know that  in the euroc_config.yaml there are two parameters that imu_topic:"/imu0" and image_topic:"/cam0/image_raw", If I change my own camera and IMU,how can I set?

highlightz, 2017-06-21:
@Dorothy-2016 First open your camera in ROS, and run command **rostopic list**, which will show the topic name of your camera image. Do the same thing to your IMU sensor. After you have got the topic names of these sensors, copy them and paste to the .yaml file.

Dorothy-2016, 2017-06-28:
OK,Thank you very much!@highlightz

trinamntn08, 2017-08-21:
Hi, 
I want to test VINS on my own dataset in mode offline. In fact, i have a folder which contains sequence images,a .csv file containing timestamps for each image,  another .csv file containing timestamps, gyroscope, accelerometers. Do you know how to put them as input for VINS-system? 
I have been blocking in this step for a long time. 
I'am really appreciate your help!

highlightz, 2017-10-11:
@trinamntn08 An easy way to feed your own dataset to VINS-system is to create ROS bag from your dataset. Refer to this page for more details please. 
[https://github.com/ethz-asl/kalibr/wiki/bag-format](https://github.com/ethz-asl/kalibr/wiki/bag-format)