HKUST-Aerial-Robotics/VINS-Mono #18: In double2vector(), why do you only use yaw_diff to update states?

highlightz, 2017-06-08:
From my understanding, in Estimator::optimization(), you update states(Rs, Ps, Vs, Bas, Bgs) after a round of optimization using Estimator::double2vector(). And there in double2vector(), you update states Ps, Vs and Rs in sliding window with a rot_diff matrix. What confuses me is the way rot_diff computed, because you just set pitch and roll to zero to pass to Utility::ypr2R().
[image]

Could I know the reason for such an operation? Looking forward to your answer.

qintony, 2017-06-08:
If you study the observability of visual-inertial system, you will know four degree-of-freedom are fully unobservable (absolute x, y, z and yaw angle). 
In our optimization, we don't lock yaw angle. So it can be solved arbitrarily (any yaw angle satisfies the cost function). So the meaning of rot_diff is to make the yaw angle smooth with the last states.

highlightz, 2017-06-08:
So, can I understand in this way that, after an optimization, yaw angle is smoothed using the delta_yaw between optimized Quaternion(para_Pose[0][6], para_Pose[0][3], para_Pose[0][4], para_Pose[0][6]) and original Rs[0], while the other parts in Rs[0] which are related to pitch and roll just keep the results from the optimization process? And the same rule works for Ps and Vs?

qintony, 2017-06-08:
Yes. right.

SkyRiderMike, 2023-09-13:
why not smooth absolute xyz in the same way ?