HKUST-Aerial-Robotics/VINS-Mono #19: The problem of MH_05_difficult result

ChenJia2017, 2017-06-09:
This is my running result of MH_05_difficult dateset,maybe there is something wrong.I did not make any changes to your code. Could you please help me do an analysis ?Thanks!
[image]

qintony, 2017-06-09:
If you did not make any changes to my code, please try another computer...
To be honest, I don't know the reason from only one figure.

ChenJia2017, 2017-06-09:
I have just updated your latest code.and it runs well.Excellent job!and ,What does the pink line represent, the loop closure point?Thanks! 
[image]

qintony, 2017-06-09:
loop links. Connect two frames which are similar.

ChenJia2017, 2017-06-09:
I just tested it again,the first error result was generated in the bebug mode of virtual maching.The error may be  related to the performance of  a computer.

ChenJia2017, 2017-06-09:
I just test it again, the first error result was generated in the debug mode of a virtual machine.I guess may be the computer perfomance will affect the results.

qintony, 2017-06-09:
VINS_mono needs some computation resource. The virtual machine is too weak.

IQ17, 2017-08-03:
Hi, you mentioned that VINS_mono needs some computional resourse, how about VINS mobile? Do they have the same performance, or VINS mobile needs less computational resourses?

qintony, 2017-08-03:
The same algorithm.
The VINS_mobile is a little simplified version of VINS_mono.
VINS_mobile need less computational resourses.

Weirdo-p, 2021-11-22:
> I just test it again, the first error result was generated in the debug mode of a virtual machine.I guess may be the computer perfomance will affect the results.

Hi, I also met the same problem since I used the debug mode (on a normal computer) in CMakeLists.txt like this:
```
set(CMAKE_BUILD_TYPE "Debug")
set(CMAKE_CXX_FLAGS "-std=c++11")
#-DEIGEN_USE_MKL_ALL")
# set(CMAKE_CXX_FLAGS_DEBUG "-O3 -Wall -g")
```
But the bug was fixed when the last command was uncommented which could be inconvenient to debug.
```
set(CMAKE_BUILD_TYPE "Debug")
set(CMAKE_CXX_FLAGS "-std=c++11")
#-DEIGEN_USE_MKL_ALL")
set(CMAKE_CXX_FLAGS_DEBUG "-O3 -Wall -g")
```
Do you have any advices? Thank you in advance. The Visual Studio Code is used as IDE.