HKUST-Aerial-Robotics/VINS-Mono #20: No trajectory display on rviz gui.

djw8899, 2017-06-10:
I follow the build steps of VINS-Mono on ROS.
When I play MH_05_difficult.bag, there is no trajectory displaying on rviz gui, just like this:

[image]

The result remains the same on MH_x_xxx.bag & VX_X_XXX.bag.
And I found that many people met the same problem,
for example https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/11

Would you mind giving some advice?

qintony, 2017-06-10:
What's the output in the "euroc.launch" terminal?

djw8899, 2017-06-10:
roslaunch vins_estimator euroc.launch 
roslaunch vins_estimator vins_rviz.launch
rosbag play YOUR_PATH_TO_DATASET/MH_05_difficult.bag 

After I input the 3rd command, "euroc.launch" terminal print these content:
vins_estimator: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 270; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[vins_estimator-3] process has died [pid 2664, exit code -6, cmd /home/worker/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/worker/.ros/log/3c1b515a-4dee-11e7-b907-000c293b7ede/vins_estimator-3.log].
log file: /home/worker/.ros/log/3c1b515a-4dee-11e7-b907-000c293b7ede/vins_estimator-3*.log

Sorry for my carelessness,  should notice this earlier..

djw8899, 2017-06-10:
Besides upstairs content, there is no more useful information in log files.
According to http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html,
maybe should use gdb to locate crash code which use Eigen.

qintony, 2017-06-11:
Your problem is same with #15, not the RVIZ issue.
The possible reason is Eigen version.
I have asked #15 to check the Eigen, it seems doesn't help. I am not sure.
If you figure it out, please tell me. 
Thanks!

djw8899, 2017-06-11:
I switch the Eigen version from 3.3.x to 3.2.0-8
The result is the same as https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/15

djw8899, 2017-06-11:
Could you provide your Eigen .deb file on your platform for us ?

qintony, 2017-06-11:
I install Eigen with apt-get .
My workspace is very clean with only one version of Eigen  3.2.0 .
I haven't installed Eigen 3.3 or build something based on Eigen 3.3.

djw8899, 2017-06-12:
When you install Eigen with ap-get, it will store the .deb file in path:
/var/cache/apt/archives/
there you input "ls -l | grep eigen", it should has some eigen related .deb file.

qintony, 2017-06-12:
[image]
no deb file related with Eigen..

I have tried with Eigen 3.3.3. The VINS_mono can run successfully with Eigen 3.3.3 ...

mlyarthur, 2017-06-12:
do you konw how to locate the ceres while using findpackage(ceres required) in cmakelists.I doubt that i uses the wrong ceres built by eigen 3.3.3

qintony, 2017-06-12:
set(EIGEN_INCLUDE_DIR "/usr/include/eigen3") or "/usr/local/include/eigen3", depends on your version
include_directories(
  ${EIGEN3_INCLUDE_DIR}
)

@mlyarthur

mlyarthur, 2017-06-12:
have you solve the problem? @djw8899

djw8899, 2017-06-12:
@mlyarthur 
According to qintony's feedback, maybe it is not the version of Eigen causes the issue.
My friend setup the environment on Ubuntu 16.04 64bit, and everything works fine...
So I decided to work on 16.04 too...

mlyarthur, 2017-06-12:
I work on the Ubuntu 14.04 32bit ,what is yours? @djw8899

djw8899, 2017-06-12:
@mlyarthur 
the same to you ...

wangquan102, 2017-06-14:
the output in the "euroc.launch" terminal 
[image]

Where the problem may lie?

qintony, 2017-06-14:
@wangquan102 Eigen version or Ceres version?

wangquan102, 2017-06-14:
@qintony  Ubuntu 16.04  ros kinetic  
  Eigen :3.2.92
ceres: 1.12.0

qintony, 2017-06-14:
@wangquan102 which bag are you trying? or this issue appears on all the bag?.

wangquan102, 2017-06-14:
@qintony i run it with my camera and imu not bag files

Dorothy-2016, 2017-07-10:
@wangquan102 I have the same question when I run this linux code with my own camera and imu,Have you solved it?

caoxudong0513, 2017-07-13:
@wangquan102 @Dorothy-2016  I have the same question when I run this linux code with my own camera and  @imu，too,Have you solved it?

MasterIzumi, 2017-12-13:
Well... I met with this problem and solved it by chance. The reason was about the versions of Eigen and Ceres in my system. So I solved it by following steps:
1. re-install Eigen.
2. re-install Ceres-solver.
3. re-compile VINS package.
Trajectories can be shown in RVIZ after I finish these steps.
However, I'm not sure that my solution can solve all these problems...

P.S. My system is Ubuntu 14.04 64bit + indigo

suraj-maniyar, 2018-04-12:
Hi,
I could run the code on dataset MH01_Easy.

 I am facing issues while running VINS-Mono on real time data from unsynced IMU (Bosch bno055)and rolling shutter camera (raspicam) published from raspberry pi. I changed the topic names in the euroc_config_no_extrinsic.yaml file according to my published topic. I am able to obtain a very sparse point cloud along with tracked key-points(Blue) and rare occurrences of matched points(Red). 

I am getting the following message on VINS Mono terminal :
[image]

The following is the screen-shot of the vins-estimator in rviz:
[image]

I am unable to obtain the localization as well as the IMU data on RVIZ due to above error. 
I am referring the following code for publishing IMU data: https://github.com/adafruit/Adafruit_Python_BNO055.

I am not sure of the exact error. 

I am using Ubuntu 16.04 with ROS Kinetic.

Yannnnnnnnnnnn, 2018-08-02:
Thx, I also met this problem.  
I found out the reason is that there were two eigen3 in my system. So I delete one of the them and rebuild the ceres and vins-mono, everything is okay NOW!
[image]

witignite, 2019-11-20:
@Yannnnnnnnnnnn Could you please let me know what version of your `Eigen` is? Did you build the library by yourself or install with `sudo apt install libeigen3-dev`?

QiruiGu, 2019-12-06:
> Thx, I also met this problem.
> I found out the reason is that there were two eigen3 in my system. So I delete one of the them and rebuild the ceres and vins-mono, everything is okay NOW!
> [image]

How you rebuild your ceres? Please...

zhaoworking, 2020-11-20:
> Hi,
> I could run the code on dataset MH01_Easy.
>
> I am facing issues while running VINS-Mono on real time data from unsynced IMU (Bosch bno055)and rolling shutter camera (raspicam) published from [...]

Same problem , but with my own dataset derived from drone fly.