HKUST-Aerial-Robotics/VINS-Mono #21: ar-box.bag file. Outdoor version.

djw8899, 2017-06-12:
http://www.bilibili.com/video/av10813089/
In this video, there is indoor and outdoor version.

But in github main page, ar-box.bag file is only indoor version.
Could you provide outdoor version too? Thanks!

qintony, 2017-06-12:
To be honest, we don't have available space to storage such big file (indoor 10G, outdoor 40G).

We usually put bags on Dropbox. If you can provide a free space (available for mainland China), I am happy to share the bag with you.

djw8899, 2017-06-12:
http://pan.baidu.com can provide more than 2T space for free and available for mainland China

qintony, 2017-06-12:
https://pan.baidu.com/s/1miv8Yhi
@djw8899 
The outdoor ar_demo. The procedure is same with the indoor ar_demo. Just play this bag file. You don't need to change other files.

djw8899, 2017-06-12:
Thanks a lot !

closed 2017-06-12 by djw8899