HKUST-Aerial-Robotics/VINS-Mono #25: Question about timing statistics of your VINS framework

highlightz, 2017-06-16:
Qin,
My github corrupts, so I have to contact you by email. I notice in paper Monocular Visual-Inertial State Estimation for Mobile Augmented Reality(ISMAR) that on an iPhone device, the VINS runs with below timing statistics(Table 1):
[image]

Herein, Loop detection in thread 3 costs 60 ms.

On the other hand, in your another paper, i.e. VINS-Mono: A robust and Versatile Monocular Visual-Inertial State Estimator（TRO）, I see that on an Inter i7 platform, the timing statistics behaves as below:
[image]
Here, in thread 3, the loop detection time becomes 100 ms. I cannot understand why a cellphone diveice has a better performance in timing than a powerful PC?

Look forward to your answer.

qintony, 2017-06-16:
The loop detection time is linear with respect to the database size. It's very straightforward, we need to compare with every keyframe in the database. Also, the feature number also influence the time. We use more features on the PC, while fewer features in the mobile.

In the mobile platform, we maintain the hundred-level database. (400)

In the PC platform, we maintain the thousand-level database, as mention in the TRO paper. (2000)

The PC detects loop at least 3 times faster than mobile for every single comparison.

highlightz, 2017-06-16:
got that. Thanks a lot.

groundmelon, 2017-06-17:
Hi, 高明

groundmelon, 2017-06-17:
@highlightz Hi，高明

highlightz, 2017-06-23:
@groundmelon Tianbo, world is small. Haha.(Chinese input method corruped)