HKUST-Aerial-Robotics/VINS-Mono #26: Changed with my own topic, but it report的

kevinchow1993, 2017-06-16:


qintony, 2017-06-16:
set the right image_width and image_height in config file.

kevinchow1993, 2017-06-16:
Thanks, **My image is 640*480**, and I set  image_width=640, image_height=480. It turned to be wrong.  but I set 752*480 just now,** it worked!!** 
Do you know why?

qintony, 2017-07-02:
are you using fisheye? if you use fisheye, please draw a cirle mask in the same size.

trinamntn08, 2017-08-31:
@kevinchow1993 i have the same issue but it didn't change when i modified image_width and image_height. Do you know why  or you modified something else ?

chrismath, 2018-10-01:
@yueshukun  I also go the same error 
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/core/src/matrix.cpp:492: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat
if you have solved it please could you tell me

karnikram, 2019-05-30:
If you're using any of the kitti raw sync sequences, you need to adjust the image size accordingly in the config file. The images are cropped to account for the undistortion effects making the size smaller than the original size of 1392 x 512.

nopnop2002, 2023-02-17:
Change image_width & image_height depend on your camera.

```
vi ./catkin_ws/src/VINS-Mono/config/euroc/euroc_config.yaml
camera_name: camera
#image_width: 752
image_height: 480
image_width: 640
```