HKUST-Aerial-Robotics/VINS-Mono #27: question about the imu frequency

shibowing, 2017-06-16:
hello, i have some problems about testing the project on device when i launch the euroc_no_extrinsic_param.launch. i use my own imu and rgbd camera. i can see the tracked feature with red points. However , i can not see the pose estimation result and the terminal always remind the "IMU excitation not enouth!". i had calibrate my camera and changed the topic of imu and  camera. However, i can not see the tf message. I guess i move the camera without acceleration.i accelerate the camera and then the terminal remind me "unstable features tracking, please slowly move you device!".    The frequency of imu is 50HZ and i did not use the static tf to attach imu with camera. Is that the reason causing faliure?  Thans for your advices
   

[image]
warning:[ INFO] [1497605197.150036705]: misalign visual structure with IMU
[ INFO] [1497605197.150427043]: IMU excitation not enouth!
[ WARN] [1497605197.271706283]: gyroscope bias initial calibration  0.0121866  0.0426114 0.00346873
[ INFO] [1497605197.273078185]: misalign visual structure with IMU
[ INFO] [1497605197.273471374]: IMU excitation not enouth!
[ WARN] [1497605197.374822350]: gyroscope bias initial calibration 0.000691495     0.03545 -0.00651129
[ INFO] [1497605197.375756768]: misalign visual structure with IMU
[ INFO] [1497605197.376261493]: IMU excitation not enouth!

PeiliangLi, 2017-06-16:
50HZ IMU is not enough for a good estimation result, please use at least 100HZ if possible.
And another possible reason is the imu camera sync problem since there always log "misalign visual structure with IMU".

shibowing, 2017-06-17:
hi, i am a beginner of ros .  Thanks for your advice.  I know how to increase the frequency of imu. However, i have puzzled about how to synchronize the imu  and camera. 
Actually, i have a xtion camera and xsens imu.  I just launched the xsens driver and openni driver at the same time.  Is that  a way of synchronization?  Can you tell me a way to synchronize the imu and camera? Thanks for your advice

caoxudong0513, 2017-07-13:
@shibowing i have the same problem with you ,Have you solved it?

qintony, 2017-07-14:
@shibowing you should make sure the IMU outputs cover gravity (the norm near 9.8, instead of 1).
As I mentioned in the README, it's not easy to have a good performance by directly combining the two unsynchronized sensors. If you can get the timestamp from the bottom driver, the timestamp is roughly ok. If you give the timestamp by yourself when you reading this message, it's always too late, maybe more than 100ms latency already happens.

For 30 HZ image, the time interval between two images is 33ms, if your IMU comes later more than 33ms, the IMUs between two images is totally wrong. How can VINS work?
The time offset at least under 1/3 interval between two images.

wangliuliu, 2017-09-11:
@qintony hi,I have some problem about IMU datas , how can I get IMU outputs cover gravity?I use NGIMU ,and get the Accelerometer axis in g whose norm always near 1.

qintony, 2017-10-09:
manually multiply the norm of gravity. @wangliuliu

SubMishMar, 2018-05-01:
Hi, I am also getting a warning that the visual structure is misaligned with IMU, my IMU units are okay and in rest it echoes the correct gravity vector, yet I keep getting this error. So I think it might be a synchronization issue, how can I know if the IMU is well synchronized with my camera ?

FYI, I am using a pixhawk with px4 firmware which publishes the imu data at 245 Hz(approximately)
and a chameleon 3 pointgrey camera which publishes images at 30 Hz(approximately). The frequencies are not constant and hover (slowly) around the above numbers.