HKUST-Aerial-Robotics/VINS-Mono #29: test on intel euclid

tongtybj, 2017-06-19:
Hi!
We are testing the vins-mono on the [intel euclid](https://click.intel.com/intelr-euclidtm-development-kit.html), of which the cpu is Intel® Atom™ x7-8700 Quad-Core.

When we run the sample launch file: ```     roslaunch vins_estimator euroc.launch ```, we found the odometry
only updated at 5Hz.

Then we also tried the vins-estimator with the onboard imu and pinhole camera, but we could not get the convergent result(of course we modified the yaml file for the basic parameter).

Any suggestion about the low frequency and bad vins result?

tongtybj, 2017-06-19:
The configuration of sensors arrangement:

[image]

tongtybj, 2017-06-19:
Config(yaml) for vins_mono:

```
File Edit Options Buffers Tools YASnippet Help                                                                                                                                                                                                                                                                     
%YAML:1.0

#common parameters
imu_topic: "/camera/imu/filtered"
image_topic: "/camera/color/image_raw"
output_path: "/config/euclid.csv"         # vins outputs will be wrttento vins_folder_path + output_path

#camera calibration
model_type: PINHOLE
camera_name: camera
image_width: 640
image_height: 480
distortion_parameters:
   k1: 0.05669641122221947
   k2: -0.11060512065887451
   p1: 7.036590250208974e-05
   p2: -0.0015619343612343073
projection_parameters:
   fx: 615.6002197265625
   fy: 617.1616821289062
   cx: 321.99688720703125
   cy: 243.24278259277344

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 1   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
                        # 2  Don't know anything about extrinsic parameters. You don't need to give R,T. We will try to calibrate it. Do some rotation movement at beginning.
ex_calib_result_path: "/config/euroc/ex_calib_result.yaml"  # If you choose 1 or 2, the extrinsic calibration result will be written vins_folder_path + ex_calib_result_path.
#If you choose 0 or 1, you should write down the following matrix.
#Rotation from camera frame to imu frame, imu^R_cam
extrinsicRotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [1, 0, 0,
           0, 1, 0,
           0, 0, 1]
#Translation from camera frame to imu frame, imu^T_cam
extrinsicTranslation: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [0.0, 0, 0.0]

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
equalize: 1             # if image is too dark or light, trun on equalize to find enough features
fisheye: 0              # if using fisheye, trun on it. A circle mask will be loaded to remove edge noisy points

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.2          # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.02         # gyroscope measurement noise standard deviation.     #0.05
acc_w: 0.0002         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.81007     # gravity magnitude

#loop closure parameters
loop_closure: 0   #if you want to use loop closure to minimize the drift, set loop_closure true and give your brief pattern file path and vocabulary file path accordingly;
                     #also give the camera calibration file same as feature_tracker node
pattern_file: "/support_files/brief_pattern.yml"
voc_file: "/support_files/brief_k10L6.bin"
min_loop_num: 25

```

qintony, 2017-06-19:
Low frequency is caused by insufficient computation power.

tongtybj, 2017-06-19:
any suggestion to modify parameters for those kind of pc.

wangquan102, 2017-06-20:
hi, tongtybj  how do you solve the  imu camera sync problem of Euclid , and where is the topic "/camera/imu/filtered" come from?  since  I got a warning "misalign visual structure with IMU"  thanks !

SueeH, 2017-06-21:
what about the performance with your device ？i have a similar device and a poor performance, the trajectory always run far while my device move little。