HKUST-Aerial-Robotics/VINS-Mono #31: cannot launch feature_tracker

SunggooJung, 2017-06-20:
Hi, thank you for your valuable open-source code.

By the way, I cannot launch the feature_tracker packge.

When I launch >> roslaunch vins_estimator euroc.launch 
------------------------------------------------------------------------------------------------------------------------------------
ubuntu@tegra-ubuntu:~/catkin_ws$ roslaunch vins_estimator euroc.launch 
... logging to /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/roslaunch-tegra-ubuntu-10137.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:39893/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/ubuntu/catk...
 * /feature_tracker/vins_folder: /home/ubuntu/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /vins_estimator/config_file: /home/ubuntu/catk...
 * /vins_estimator/vins_folder: /home/ubuntu/catk...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    vins_estimator (vins_estimator/vins_estimator)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[feature_tracker-1]: started with pid [10155]
process[vins_estimator-2]: started with pid [10156]
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp, line 863
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp:863: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

[ INFO] [1497969209.836649578]: init begins
[feature_tracker-1] process has died [pid 10155, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/feature_tracker-1.log].
log file: /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/feature_tracker-1*.log
[ INFO] [1497969209.853653828]: Loaded config_file: /home/ubuntu/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
[ INFO] [1497969209.855908798]: Loaded vins_folder: /home/ubuntu/catkin_ws/src/VINS-Mono/feature_tracker/../config/../
[ WARN] [1497969209.856349729]:  Optimize extrinsic param around initial guess!
[ INFO] [1497969209.857591900]: Extrinsic_R : 
2.22045e-16          -1           0
          1 2.22045e-16           0
          0           0           1
[ INFO] [1497969209.857880906]: Extrinsic_T : 
-0.02 -0.06  0.01
[ WARN] [1497969209.858156528]: waiting for image and imu...
[ WARN] [1497969209.917905097]: DATA SUBSCRIBED
[ WARN] [1497969209.918072907]: LOOP_CLOSURE true
voc file: /home/ubuntu/catkin_ws/src/VINS-Mono/feature_tracker/../config/..//support_files/brief_k10L6.bin
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp, line 863
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp:863: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

[vins_estimator-2] process has died [pid 10156, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/vins_estimator-2.log].
log file: /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/vins_estimator-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
------------------------------------------------------------------------------------------------------------------------------------
Therefore I try to launch feature_tracker directly to see the error part, it shows

------------------------------------------------------------------------------------------------------------------------------------
ubuntu@tegra-ubuntu:~/catkin_ws$ roslaunch feature_tracker euroc.launch 
... logging to /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/roslaunch-tegra-ubuntu-9993.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:45356/

SUMMARY
========

PARAMETERS
 * /feature_tracker/F_threshold: 1.0
 * /feature_tracker/calib_dir: /home/ubuntu/catk...
 * /feature_tracker/cam_name0: euroc0_camera_cal...
 * /feature_tracker/equalize: False
 * /feature_tracker/fisheye: False
 * /feature_tracker/freq: 1
 * /feature_tracker/max_cnt: 200
 * /feature_tracker/min_dist: 20
 * /feature_tracker/show_track: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    player (rosbag/play)

auto-starting new master
process[master]: started with pid [10004]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 00587f2e-55c5-11e7-a1fc-00044b57fe0c
process[rosout-1]: started with pid [10017]
started core service [/rosout]
process[player-2]: started with pid [10024]
process[feature_tracker-3]: started with pid [10033]
[ERROR] [1497969027.609730521]: Failed to load config_file
OpenCV Error: Null pointer (NULL or empty buffer) in cvOpenFileStorage, file /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp, line 4153
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.2.0/modules/core/src/persistence.cpp:4153: error: (-27) NULL or empty buffer in function cvOpenFileStorage

[feature_tracker-3] process has died [pid 10033, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/feature_tracker/feature_tracker ~imu:=/imu0 ~raw_image:=/cam0/image_raw __name:=feature_tracker __log:=/home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/feature_tracker-3.log].
log file: /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/feature_tracker-3*.log
[player-2] process has finished cleanly
log file: /home/ubuntu/.ros/log/00587f2e-55c5-11e7-a1fc-00044b57fe0c/player-2*.log
------------------------------------------------------------------------------------------------------------------------------------

Actually I still not understood the meaning of below two list, in the feature_tracker/launch/euroc.launch 
        
        <param name="calib_dir" type="string" value="$(find feature_tracker)/config/" />
        <param name="cam_name0" type="string" value="euroc0_camera_calib_mei.yaml" />

Thanks.

qintony, 2017-06-20:
We don't maintain feature_tracker/euroc.launch for a long time, it can't be used at all. Sorry for that, we will remove this launch file in the next version.
We develop VINS_mono based on Opencv 2.4. This issue may be caused by Opencv 3.2. 
From the print information, it can be seen the problem exists in the file interface when loading the file by Opencv.
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName
It seems to be easy to solve it by yourself.

duyilong, 2017-06-26:
@qintony 
execuse me, I feel confused about your explanation here, can I get more details about it?
I run this project(latest version till today) on my PC(Ubuntu 14.04 + ROS Indigo + OpenCV2.4.8(installed with Indigo), no extra opencv release version), when I launch

 ### `roslaunch vins_estimator euroc.launch`

 I encounter the same OpenCV Error.

However I can't get the proper solutions to this Error. It would be kind of you to show me more detalis about it, thanks very much.

qintony, 2017-06-27:
@duyilong are your Ubuntu is 64 bit?
Can you paste your terminal output here?

duyilong, 2017-06-27:
@qintony 
It is Ubuntu 64bit.
here is the terminal output:

> [duyl@ ~/catkin_ws]15:31$ roslaunch vins_estimator euroc.launch
> ... logging to /home/duyl/.ros/log/be91a318-5b0a-11e7-ae9c-b191e14994a5/roslaunch-Vestro-13435.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://Vestro:33430/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /feature_tracker/config_file: /home/duyl/catkin...
>  * /feature_tracker/vins_folder: /home/duyl/catkin...
>  * /rosdistro: indigo
>  * /rosversion: 1.11.21
>  * /vins_estimator/config_file: /home/duyl/catkin...
>  * /vins_estimator/vins_folder: /home/duyl/catkin...
> 
> NODES
>   /
>     feature_tracker (feature_tracker/feature_tracker)
>     vins_estimator (vins_estimator/vins_estimator)
> 
> auto-starting new master
> process[master]: started with pid [13447]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to be91a318-5b0a-11e7-ae9c-b191e14994a5
> process[rosout-1]: started with pid [13460]
> started core service [/rosout]
> process[feature_tracker-2]: started with pid [13463]
> process[vins_estimator-3]: started with pid [13464]
> [ INFO] [1498548740.240432998]: init begins
> [ INFO] [1498548740.312212456]: Loaded config_file: /home/duyl/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
> 
> > **OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740**
> > **[ INFO] [1498548740.315501276]: Loaded vins_folder: /home/duyl/catkin_ws/src/VINS-Mono/feature_tracker/../config/../**
> > **OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740**
> > **terminate called after throwing an instance of 'cv::Exception'**
> >   **what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName**
> > 
> > **terminate called after throwing an instance of 'cv::Exception'**
> >   **what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName**
> 
> [feature_tracker-2] process has died [pid 13463, exit code -6, cmd /home/duyl/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/duyl/.ros/log/be91a318-5b0a-11e7-ae9c-b191e14994a5/feature_tracker-2.log].
> log file: /home/duyl/.ros/log/be91a318-5b0a-11e7-ae9c-b191e14994a5/feature_tracker-2*.log
> [vins_estimator-3] process has died [pid 13464, exit code -6, cmd /home/duyl/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/duyl/.ros/log/be91a318-5b0a-11e7-ae9c-b191e14994a5/vins_estimator-3.log].
> log file: /home/duyl/.ros/log/be91a318-5b0a-11e7-ae9c-b191e14994a5/vins_estimator-3*.log
> 

To be honest, I even couldn't find the _persistence.cpp_ file mentioned in the Error, which is pretty awkward. Orz...

AaboutL, 2017-06-30:
@duyilong 
I just met the same error as yours. There were opencv2.4 and opencv3.2 in my ubuntu 14.04. After removed opencv3.2, clean catkin_ws/build, and executed catkin_make, the Mono can run. If you haven't fixed this error, hope this can help.

duyilong, 2017-07-03:
@AaboutL Thanks for your advise, but that didn't work.  Same Error happened.
I run 

`pkg-config --modversion opencv`

and the output is 

> 2.4.8

does that mean I have remove opencv-2.4.11 already?

varun3108, 2020-11-03:
... logging to /home/mindbeat/.ros/log/78fc1f5e-1dcb-11eb-9532-6cc21762b5e1/roslaunch-DESKTOP-HBDV97J-5759.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://DESKTOP-HBDV97J:64590/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/mindbeat/ca...
 * /feature_tracker/vins_folder: /home/mindbeat/ca...
 * /pose_graph/config_file: /home/mindbeat/ca...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vins_estimator/config_file: /home/mindbeat/ca...
 * /vins_estimator/vins_folder: /home/mindbeat/ca...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [5769]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 78fc1f5e-1dcb-11eb-9532-6cc21762b5e1
process[rosout-1]: started with pid [5780]
started core service [/rosout]
process[feature_tracker-2]: started with pid [5786]
process[vins_estimator-3]: started with pid [5788]
process[pose_graph-4]: started with pid [5789]
[ INFO] [1604404533.602130900]: init begins
[ INFO] [1604404533.656113900]: Loaded config_file: /home/mindbeat/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /home/shaozu/output//vins_result_no_loop.csv
/home/shaozu/output/ not exists, trying to create it
/home/shaozu not exists, trying to create it
Failed to create folder /home/shaozu
[ INFO] [1604404533.693287700]: ROW: 480.000000 COL: 752.000000
[ WARN] [1604404533.694784300]:  fix extrinsic param
[ INFO] [1604404533.700555800]: Extrinsic_R :
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1604404533.722156100]: Extrinsic_T :
-0.0216401  -0.064677 0.00981073
[ INFO] [1604404533.728395500]: Synchronized sensors, fix time offset: 0
[ WARN] [1604404533.730781000]: waiting for image and imu...
vocabulary_file/home/mindbeat/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/mindbeat/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
/home/shaozu/output/pose_graph/ not exists, trying to create it
/home/shaozu/output not exists, trying to create it
/home/shaozu not exists, trying to create it
Failed to create folder /home/shaozu
/home/shaozu/output/ not exists, trying to create it
/home/shaozu not exists, trying to create it
Failed to create folder /home/shaozu
no previous pose graph
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 483
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:483: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

[feature_tracker-2] process has died [pid 5786, exit code -6, cmd /home/mindbeat/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/mindbeat/.ros/log/78fc1f5e-1dcb-11eb-9532-6cc21762b5e1/feature_tracker-2.log].
log file: /home/mindbeat/.ros/log/78fc1f5e-1dcb-11eb-9532-6cc21762b5e1/feature_tracker-2*.log