HKUST-Aerial-Robotics/VINS-Mono #33: Question about alignment of vector g with world frame?

highlightz, 2017-06-23:
I guess that at the end of visual inertial alignment(corresponding to function Estimator::visualInitialAlign), shown below:
[image]

you did the operations described as in your TRO paper:
[image]

so I guess rot_diff stands for w_q_c0 in the paper.
Am I correct? And I am a little confused about function Utility::r2R(), what is the meaning of these operations?
[image]

PeiliangLi, 2017-07-02:
Because pitch and roll are global observable.

qintony, 2017-07-02:
We just want to set the yaw angle of the first frame is zero.

highlightz, 2017-07-04:
got it.