HKUST-Aerial-Robotics/VINS-Mono #36: How to set the initial parameters of imu?

Markbess, 2017-06-27:
In your  configuration folder /config/euroc_config.yaml, here we need to configure the parameters of imu, including accelerometer measurement noise standard deviation acc_n, gyroscope measurement noise standard deviation gyr_n and gyroscope bias random work noise standard gyr_w. But  when using our own hardware(camera+imu), how to get the IMU parameters?

Here, you said, "The more accurate parameters you provide, the better performance." But I guess that these parameters will be changed via nonlinear optimization equation, and that means they are not that important just as the initial value of the optimization.  I do not know if the understanding is correct. 
[image]

PeiliangLi, 2017-07-02:
Basically, you can just use these noise parameters we provided. What's more important is the intrinsic, and time offset when you use your devices.  
But if you have extremely good and synced imu hardware, you can decrease the imu noise parameter to get better performance.

Markbess, 2017-07-02:
@PeiliangLi 
Thank you for your quick response. The hardware we used is Multisense S7 which includes a camera and a IMU. Here, I can obtain images and IMU data via ROS topic which has accurate  timestamp. But, the program does not work very well, and it warns,"Big IMU acc bias estimation... numerical unstable in pre-integration."
[image]
Breakpoint HIT!
[image]
Here, The noise parameter of IMU is set to the default you provided.
I am wondering does the hardware I used meet the your requirements?
The hardware details.
http://docs.carnegierobotics.com/S7/

Markbess, 2017-07-02:
Thank you for your quick response. The hardware we used is Multisense S7 which includes a camera and a IMU. Here, I can obtain images and IMU data via ROS topic which has accurate  timestamp.But, the program does not work very well, and

pangfumin, 2017-07-20:
@qintony 
Thank you for sharing this work !  I am also trying to use my own sensor rig to run this VINS. I found your defaut IMU noise parameters for EuRoc dataset  is somehow much  larger than what I found in Euroc imu parameter file.  I wonder how did you set the defaut parameters. Have you tune IMU noise parameters to find a set of good values?
Thank you again for your great work!

mrr-05, 2020-09-02:
Hello! I am having problems in my Gazebo simulation. I get the following messages:
failure detection!
system reboot!
I think it can be because IMU parameters of euroc_config.yaml are not correct, but i don't know how I can obtain these values.
Thank you!

MaxHonggg, 2020-09-06:
> @PeiliangLi
> Thank you for your quick response. The hardware we used is Multisense S7 which includes a camera and a IMU. Here, I can obtain images and IMU data via ROS topic which has accurate timestamp. But, the program does not work very well, and it warns,"Big IMU acc bias estimation... numerical unstable in pre-integration."
> [image]
> Breakpoint HIT!
> [image]
> Here, The noise parameter of IMU is set to the default you provided.
> we are wondering does the hardware we used meet the your requirements?
> The hardware details.
> http://docs.carnegierobotics.com/S7/
> Thank you again for everything you've done.

Hello,did you solve the problem? I am working on the running VINS on my device too, thank you~~~~