HKUST-Aerial-Robotics/VINS-Mono #37: Eigen fail in Odroid u3

tdnet12434, 2017-06-27:
Hi! Thank you for make code available!

I have successful run in my labtop but when i try to run at odroid u3, the following error shown up.

```
[ WARN] [1498578211.335453879]: have no prior about extrinsic param, calibrate extrinsic param
[ WARN] [1498578211.335672337]: waiting for image and imu...
[ WARN] [1498578211.779238376]: throw img, only should happen at the beginning
[ WARN] [1498578211.990729839]: throw img, only should happen at the beginning
[ INFO] [1498578212.191727203]: calibrating extrinsic param, rotation movement is needed
vins_estimator: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 270; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[vins_estimator-2] process has died [pid 26489, exit code -6, cmd /home/odroid/quad_ws/devel/lib/vins_estimator/vins_estimator /imu0:=/mavros/imu/data __name:=vins_estimator __log:=/home/odroid/.ros/log/f7d626c0-5b47-11e7-91c7-3a02dd67b601/vins_estimator-2.log].
log file: /home/odroid/.ros/log/f7d626c0-5b47-11e7-91c7-3a02dd67b601/vins_estimator-2*.log

```

How can i fix this? Any advice would be appreciate.
Thank you advanced!

PeiliangLi, 2017-07-02:
We don't suggest running VINS on odroid u3.
Even if you compile it successfully, we cannot make sure the performance. see #1

poaongithub, 2017-07-17:
@tdnet12434  same error on raspberry pi3, do you know why?

heguixiang, 2017-08-14:
@tdnet12434 @poaongithub , same error like you, does you resolve it?

SubMishMar, 2017-10-25:
@tdnet12434 did you fix this issue?

heguixiang, 2017-10-27:
I think I fixed it, it is a Eigen compatibility issue which in 32-bit OS. Plz see the issue #77 [link]{https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/77}.

heguixiang, 2017-10-27:
I think I fixed it, it is a Eigen compatibility issue which in 32-bit OS. Plz see the issue #77

SubMishMar, 2017-10-27:
Thanks I will take a look and let you know.