HKUST-Aerial-Robotics/VINS-Mono #38: Camera position is randomly drifting away

shaomang, 2017-06-28:
Hi, thanks for sharing the implementation. There were some issues when I testing with my own data (collected from an android mobile phone).

[image]

As shown in the screenshot, the camera randomly drifting away, usually after a camera movement but could also happen when it's almost static. 

The camera path seems very inaccurate as well, though it works fine with only camera rotation.

Since the timestamp might be inaccurate or unsynchronised, I tried few ways to fix that but still unlucky. So I'm wondering if there might be any other possible reasons for causing this?

---
The iOS version works really nice, so I'll further test it with the data collected from iphone and report the result.

shaomang, 2017-06-28:
Update: I tried with iOS data collected by myself but still not working. The iOS app version / euroc dataset works perfectly with same setting.

Markbess, 2017-06-28:
I have  encountered the same problem.

qintony, 2017-07-01:
Usually, the IMU and camera are unsynchronised on Android device, 10 - 100ms time offset, which causes the trouble.

shaomang, 2017-07-04:
Hi, thanks for the reply. I knew the problem could be unsynchronised data, so I also tried using an iPhone7+ to collect data but it still didn't work. Maybe I should post on vins-mobile page to ask if there is any tricks to synchronise.

closed 2017-07-04 by shaomang