HKUST-Aerial-Robotics/VINS-Mono #39: "Since the absolute scale, roll and pitch angle are fully observable in the visual-inertial system, accumulate drifts only occur in position (x, y, z) and yaw angle",在论文中这句话要怎吗理解？？

sujinglan123, 2017-06-29:


qintony, 2017-07-02:
You can recover absolute scale, roll and pitch in the real world from IMU.
However, you can only recover relative position and yaw angle from two or more consecutive frames.
You don't know the absolute position since every position can be set as zero. For roll and pitch angle, the gravity direction is zero, which is fixed.

pangfumin, 2017-09-27:
In visual-inertial system, the absolute scale , roll and pitch are observable. But position,and yaw are not.
If you add a delta p on all camera and landmarks, the cost function value of BA dose not change. Similar adding a delta on yaw also dose not  affect the const function. This means they are not observable.

qintony, 2017-10-09:
@pangfumin exactly!