HKUST-Aerial-Robotics/VINS-Mono #41: some doubt about code and paper

pvoid33, 2017-07-03:
I have several question about the code and paper.

1. the formula
[image]
why it just minus the noise bias without the noise. The true acceleration should be
 $$a_t = a_m + b_{at}  + n_{at}$$
2. the imu residual code
```c++
for (int i = 0; i < WINDOW_SIZE; i++)
 {
        int j = i + 1;
        if (pre_integrations[j]->sum_dt > 10.0)
            continue;
        IMUFactor* imu_factor = new IMUFactor(pre_integrations[j]);
        problem.AddResidualBlock(imu_factor, NULL, para_Pose[i], para_SpeedBias[i], para_Pose[j], para_SpeedBias[j]);
    }
```
which i think should be 
```c++
IMUFactor* imu_factor = new IMUFactor(pre_integrations[i]);
```
the `pre_integrations[i]` is the pre-intergration for image frame i to image frame i+1. Is it right? 
3, 
[image]
```c++
    Eigen::Vector3d b1, b2;
    Eigen::Vector3d a = pts_j.normalized();
    Eigen::Vector3d tmp(0, 0, 1);
    if(a == tmp)
        tmp << 1, 0, 0;
    b1 = (tmp - a * (a.transpose() * tmp)).normalized();
    b2 = a.cross(b1);
    tangent_base.block<1, 3>(0, 0) = b1.transpose();
    tangent_base.block<1, 3>(1, 0) = b2.transpose();
```
I understand the algorithmn you describe in your paper ,use two cross product to construct it. But I donot
know the math in your code , even i know i test it is right.
4, 
```c++
        if (last_marginalization_info)
        {
            vector<int> drop_set;
            for (int i = 0; i < static_cast<int>(last_marginalization_parameter_blocks.size()); i++)
            {
                if (last_marginalization_parameter_blocks[i] == para_Pose[0] ||
                    last_marginalization_parameter_blocks[i] == para_SpeedBias[0])
                    drop_set.push_back(i);
            }
            // construct new marginlization_factor
            MarginalizationFactor *marginalization_factor = new MarginalizationFactor(last_marginalization_info);
            ResidualBlockInfo *residual_block_info = new ResidualBlockInfo(marginalization_factor, NULL,
                                                                           last_marginalization_parameter_blocks,
                                                                           drop_set);

            marginalization_info->addResidualBlockInfo(residual_block_info);
        }
```
Why it will exist the code
```c++
for (int i = 0; i < static_cast<int>(last_marginalization_parameter_blocks.size()); i++)
            {
                if (last_marginalization_parameter_blocks[i] == para_Pose[0] ||
                    last_marginalization_parameter_blocks[i] == para_SpeedBias[0])
                    drop_set.push_back(i);
            }
```
which i think we have marginalize the old state. para_Pose[0] ,para_SpeedBias[0].

Thank you for your great job.

qintony, 2017-07-04:
1. We assume the noise is under  Gauss distribution, the mean value is 0.
2. 
[image]
Essentially same with the algorithm.

3. I don't know which version you are reading. The code is already 
[image]

xiaopenghuang, 2017-07-04:
```c++
 if (last_marginalization_info)
        {
            vector<int> drop_set;
            for (int i = 0; i < static_cast<int>(last_marginalization_parameter_blocks.size()); i++)
            {
                if (last_marginalization_parameter_blocks[i] == para_Pose[0] ||
                    last_marginalization_parameter_blocks[i] == para_SpeedBias[0])
                    drop_set.push_back(i);
            }
            // construct new marginlization_factor
            MarginalizationFactor *marginalization_factor = new MarginalizationFactor(last_marginalization_info);
            ResidualBlockInfo *residual_block_info = new ResidualBlockInfo(marginalization_factor, NULL,
                                                                           last_marginalization_parameter_blocks,
                                                                           drop_set);

            marginalization_info->addResidualBlockInfo(residual_block_info);
        }
```
Why we need to take care of last_marginalization_info. When we marginalzation the old frame ,I think we only need to care the imu pre_integrations and visual  start_frame is 0.
Can you provide some clue about last_marginalization_info  code.

closed 2017-07-06 by xiaopenghuang