HKUST-Aerial-Robotics/VINS-Mono #43: navigation

ZhengPan2014, 2017-07-04:
how can i replay the navigation with voxel map like in video4 ? very kind of you for any reference.

qintony, 2017-07-04:
I think @WayneTimer can help you.

WayneTimer, 2017-07-04:
The best reference is our [JFR paper](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/support_files/paper/jfr2017yi.pdf). We detail our system, module by module. I guess you can find something valuable in our paper.

ZhengPan2014, 2017-07-27:
@WayneTimer @qintony thank you so much, I get the papers at dir support files under this project. what confused me now is the scale problem(you know for pure monocular). what I see now is that IMU here is just for rotation correction, if you just reconstruct the scale from pinhole images from ROI like stereo? very kind of you for replying.

hadogenes, 2020-03-07:
Is the source code somewhere available?