HKUST-Aerial-Robotics/VINS-Mono #45: A special case which can cause diverge

xiaopenghuang, 2017-07-04:
[image]
The above picture is the imu read from pixhawk. Everything goes well before imu data surge to(35m^s-2) ,(This happens when my quadrotor hit something) ，then the odometry diverge,(in about 3 seconds, it would goes 20 meters), Which i think may need to take care of this case ,the imu read is not normal.In quadrotor it must be crash.
But why when the quadrotor hit something(imu surge just 1 second) , then it landdown(imu read normal), the odometry still drift.
[image]

qintony, 2017-07-05:
Give the bigger variance of IMU measurement.

closed 2017-07-05 by xiaopenghuang

Pallav1299, 2021-11-18:
@hxp33, can you share details about your setup? Did you do hardware time synchronization between PIXHAWK and companion computer? Since VINS already does online temporal calibration [PAPER](https://arxiv.org/pdf/1808.00692.pdf) is it actually required to do hardware synchronization?