HKUST-Aerial-Robotics/VINS-Mono #48: About hard-coded parameters

zjcv2012, 2017-07-07:
Hi,
Thanks for the amazing work. Currently, I am using this code for the initial step to generate tf for my project. One thing I noticed is that FOCAL_LENGTH is hard-coded, and it is assigned 460 in several files. If I use image data from my own camera, should I assume that this value needs to be changed?

Best.
Juncheng

qintony, 2017-07-08:
No. This parameter is just for a virtual camera projection. You don't need to change.

zjcv2012, 2017-07-10:
@qintony Hi, thanks for your quick response. I noticed that FOCAL_LENGTH appear in KeyFrame::FundmantalMatrixRANSAC, and in TemplatedLoopDetector.h as well. I just don't get why this doesn't need to be changed. Can you explain how the virtual camera projection work? Also, should I also keep the ROW and COL in TemplatedLoopDetector.h even though the actual image resolution changes?

qintony, 2017-07-10:
[image]

That's the virtual camera.

zjcv2012, 2017-07-10:
@qintony Hi, I looked in to the code you attached. It seems that the tracked point is first lifted from the image plane to a projective ray(through liftProjective()), and then is projected to a virtual image plane(probably this is what you said as virtual camera projection?). What I am not sure about is that whether the value of FOCAL_LENGTH matters or not? Since the COL and ROW in the attached code are the actual image width and height, could the FOCAL_LENGTH there be changed to the actual focal length(I did some experiments, but it seems the value of FOCAL_LENGTH does influence the performance).
One more thing, does the FOCAL_LENGTH in the loop_closure have the same function? Should that be alway kept as 460?(My test showed that keeping it as 460 had better result than changing it to the actual focal length).
Thanks in advance.

qintony, 2017-07-11:
The focal_length in loop_closure have the same function.
In fact, it's just a parameter to plot the undistorted image. It doesn't influence the performance much. 
The more accurate focal length you use, your undistorted image will be closer to the real world.
But we just use the undistorted image for a fundamental matrix test to reject outliers instead of visualization. This value doesn't influence fundamental matrix test much.

zjcv2012, 2017-07-12:
@qintony  Hi, I still have some questions.
1. In liftProjective(), the distortion() function is called. Is the distortion() for distorting the image or undistorting it. The comment for this function says "apply distortion to input point from the normalised plane", but if this is for distortion ,where does the undistortion happen?
2. Why all points after KLT tracker are first lifted to normalized plane and then projected onto a virtual image plane(just like the code you attached)? Why don't do fundamental matrix test directly since the correspondences of tracked points are known?
3. After I change the FOCAL_LENGTH from 460 to the actual focal length of my camera, the system will fail. Can you try to explain that?
Sorry for asking so many questions, I am just confused about the code and trying to understand every detail.
Thanks.

zjcv2012, 2017-07-13:
@qintony Just want you to know that I figured out answers to my questions 1 and 2 yesterday. Still thinking about question 3 though.

qintony, 2017-07-14:
1. in liftprojective(). we want to perform undistortion. however, undistortion is usually nonlinear, and hard to be written in closed form solution. so the inversely iteratively distortion function is used to perform undistortion in camera model code. It's a trick to realize undistortion.
2. The physical meaning of the KLT parameters is in the pixel. we need to match this physical meaning. so project it to the virtual camera.
3. For example, we tolerate 3-pixel noise under 460 focal lengths. If you change to 920, the tolerate pixel will be 6 pixels, since you project the point to a further plane. Some parameters should change will focal length.

zjcv2012, 2017-07-25:
@qintony Hi Tong, do you know what is the unit for the acc_w(accelerometer bias random work noise standard deviation) and the gyr_w(gyroscope bias random work noise standard deviation) in the yaml file. Are the units m/s^2/sqrt(s) and rad/s/sqrt(s)? I just want to make sure that I give the right values as input.

Thanks in advance.

closed 2017-08-03 by zjcv2012

harryHR, 2019-02-22:
@qintonguav ,Hi, why don't you directly use the undistortion functions in Opencv such as undistortPoints(),initUndistortRectifyMap() ?