HKUST-Aerial-Robotics/VINS-Mono #51: Can't find explicit PnP computation before the optimization?

highlightz, 2017-07-11:
In initial stage, there is an explicit process where PnP is performed to calculate poses of frames in window. However, in a normal nonlinear process pipeline, there seems nowhere to find such a PnP process to get newly coming frame's pose, instead, the code seem to directly enters optimization process. Where can I find the handling of newest frame's pose calculation?

highlightz, 2017-07-11:
void Estimator::solveOdometry()
{
      if (frame_count < WINDOW_SIZE)
          return;
      if (solver_flag == NON_LINEAR)
      {
          TicToc t_tri;
          f_manager.triangulate(Ps, tic, ric);
          ROS_DEBUG("triangulation costs %f", t_tri.toc());
          optimization();
      }
}

PeiliangLi, 2017-07-11:
In nonlinear part, the latest pose is propagated with imu pre-integration.

highlightz, 2017-07-11:
Could you please show me the corresponding code part? Thanks a lot.

PeiliangLi, 2017-07-11:
[image]

highlightz, 2017-07-11:
Got it. Thank you Li.