HKUST-Aerial-Robotics/VINS-Mono #53: How to test on my dataset ?

trinamntn08, 2017-07-18:
I want to test on my own dataset, but i don't know how to do that. 
In fact, i have a set of images which is named  (001 ,002 ........)  and a set of IMU data ( timestamps+acce + gyro ).  
My goal is to create an ROS bag , combining all data  to run on VINS but i don't know how to fusion these two sets of data. Can you detail how to do that, pls ?
Thanks in advance!

xuchen27, 2017-07-19:
I think you can replay  EuRoC MAV Dataset bag and see the topic.

closed 2017-08-23 by trinamntn08

zxp771, 2018-03-25:
@trinamntn08 
Hi 
Have you test your dataset by vins?
I have the same condition(I record a video and cut it into images and I record the imu measurement by a software on my phone) I also know you use the kalibr to creat the rosbag.
but the rosbag what I created can't work for the test.
could please tell some detail about the creation of your dataset to me?

Thanks for your help.

lilpeng-xxxl, 2022-10-19:
> @trinamntn08 Hi Have you test your dataset by vins? I have the same condition(I record a video and cut it into images and I record the imu measurement by a software on my phone) I also know you use [...]

The rosbag what I created can work but there is  no locus ."Not enough features or parallax;Move device around","unstable features tracking,please slowly move you device"
Have you solve your problem?
Thanks for your help!