HKUST-Aerial-Robotics/VINS-Mono #54: Big IMU acc bias estimation!

Markbess, 2017-07-19:
Why the program always shows that Big IMU acc bias estimation even if  we hold the hardware by hand and move slowly?

qintony, 2017-08-10:
Usually, that a wrong estimation caused by unsynchronization.

PatricWang, 2018-05-08:
I got the same problem but my device was in a car, the problem happened when the car started moving,
I wonder is this also caused by unsynchronization or the motion was so aggressive
@qintonguav

siyuada, 2018-07-18:
I got the same problem as @PatricWang ,happened when the car started moving. Maybe sometimes vinsmono can overcome this,played well,sometimes it just failed.
Have you got any good solution?

liuxiao916, 2022-08-29:
Hi, I solve this problem by check my rotation matrix between camera and IMU. Please read the example 2 and 3 in [extrinsic_parameter_example.pdf](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/config/extrinsic_parameter_example.pdf). Make sure you give the VINS the right extrinsic matrix.