HKUST-Aerial-Robotics/VINS-Mono #60: Successfully running demo with the latest stable version of Ubuntu, ROS, OpenCV and Eigen

marooncn, 2017-07-27:
I successfully run the demo with the latest stable version of Ubuntu, ROS, OpenCV and Eigen.
Ubuntu: 16.04 LTS
ROS: Kinetic 
OpenCV: 3.2.0
Eigen: 3.3.4
So if you have the same latest version, you can direct run with the demo.

maciejmatuszak, 2017-07-27:
Can you share your hardware specs? Processor type, number of cores and clock?
I can run it OK on i7 laptop as well.

marooncn, 2017-07-28:
Yes, @maciejmatuszak . My laptop's configuration lists below.
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 158
Model name:            Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
Stepping:              9
CPU MHz:               984.033
CPU max MHz:           3800.0000
CPU min MHz:           800.0000
BogoMIPS:              5600.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              6144K
NUMA node0 CPU(s):     0-7
But I think a lower configuration  such as an i5 CPU  is OK, too.

maciejmatuszak, 2017-07-28:
Yes, I think it was reported it run on low i5 ok as well. I am trying to get it working on Nvidia TX2, no luck so far.

closed 2017-08-10 by marooncn