HKUST-Aerial-Robotics/VINS-Mono #61: Why does the VINS_Mono have different performace int the same scene (using the same bagfile)?

xiaofeifei1234, 2017-07-27:
we use the VINS_Mono to test in the same scene however its performance depends on the computer .
First time we use the Inter NUC mini computer the performance is very bad ,the second time we use laptop the result is ok,the third time we use desktop the performance is good. So we are curious about this question that why the VINS_Mono's performance depends on the computer. Thank you!

maciejmatuszak, 2017-07-27:
I think I am dealing with the same problem. I recorded a bag file and it works on Laptop ok but try to run it on TX1, TX2 the position will run away. 
Part of the SLAM solution is Ceres solver that uses iterative approach. It need number of iterations to get good results. In software it is caped at 9 iterations or specific time. If you run it on fast machine it will reach 9 iterations and results are well optimised. On slower machine it may do 2-3 iteration before the operation is caped by time and position drifts wildly...  I am trying to figure-out why the Ceres solver takes so long on Nvidia TX1 / TX2 to the point where it drifts. It was reported bu author to run on TX1 with earlier ROS (indigo) version. See #1.

xiaofeifei1234, 2017-07-28:
Got it,thank you for prompt reply

bygreencn, 2018-08-28:
My extended question may be how to calibrate the custom device to run VINS-Mono, maybe give a detail checklist. I had tested cartographer, LOAM,LSD and DSO. At my status, I hope to combine the slam and the base map to give relative pose at the global sense.