HKUST-Aerial-Robotics/VINS-Mono #62: misalign visual structure with IMU

wangminhang, 2017-07-28:
I run VINS with RealSense ZR300 camera, and it behaviors well at most time, but I also meet the question below, and rviz has no track and pointcloud dispaly, but the featured map display well. what happened? it seems can't recover any more.
the version of my eigen is 3.2.0.

``[ INFO] [1501220302.096056537]: misalign visual structure with IMU
[ WARN] [1501220302.188962342]: gyroscope bias initial calibration  0.00499007 -0.00593529  0.00132741
[ INFO] [1501220302.199975535]: misalign visual structure with IMU
[ WARN] [1501220302.308324242]: gyroscope bias initial calibration 6.08718e-05   0.0451012  -0.0112661
[ INFO] [1501220302.320570932]: misalign visual structure with IMU
[ WARN] [1501220302.481384991]: gyroscope bias initial calibration 0.00292979  0.0152145 0.00380104
[ INFO] [1501220302.493172674]: misalign visual structure with IMU
[ WARN] [1501220302.621653798]: gyroscope bias initial calibration -0.0163037  0.0281737 -0.0091501
[ INFO] [1501220302.633536271]: misalign visual structure with IMU
[ WARN] [1501220302.751830691]: gyroscope bias initial calibration  0.0124301  0.0036569 0.00677385
[ INFO] [1501220302.765528648]: misalign visual structure with IMU
[ WARN] [1501220302.923701241]: gyroscope bias initial calibration  0.00286491   0.0281394 -0.00860164
[ INFO] [1501220302.933657255]: misalign visual structure with IMU
W0728 13:38:23.023864 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.024817 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.025691 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
[ WARN] [1501220303.028806228]: gyroscope bias initial calibration -0.0093362   0.104164 0.00744695
[ INFO] [1501220303.032728882]: misalign visual structure with IMU
[ WARN] [1501220303.169064788]: gyroscope bias initial calibration 0.00228783 -0.0244078 0.00504458
[ INFO] [1501220303.172414428]: misalign visual structure with IMU
[ WARN] [1501220303.314513588]: gyroscope bias initial calibration 0.00402812 -0.0229598 -0.0011767
[ INFO] [1501220303.318397571]: misalign visual structure with IMU
W0728 13:38:23.409066 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.411484 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.412395 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.414939 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.417507 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.418406 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.420831 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.423243 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.425581 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.428148 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.428982 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.431447 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.434072 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0728 13:38:23.435039 17002 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
[ WARN] [1501220303.442229292]: gyroscope bias initial calibration -0.000865684  -0.00272036   -0.0139762
[ INFO] [1501220303.446969226]: misalign visual structure with IMU
[ WARN] [1501220303.555974311]: gyroscope bias initial calibration -0.00127629  -0.0449397   0.0159189
[ INFO] [1501220303.561865879]: misalign visual structure with IMU
[ WARN] [1501220303.752329499]: gyroscope bias initial calibration -0.00116859  0.00650857 -0.00148748
[ INFO] [1501220303.759284553]: misalign visual structure with IMU
[ WARN] [1501220303.867266921]: gyroscope bias initial calibration 0.000755303  0.00433638 -0.00511009
[ INFO] [1501220303.873092236]: misalign visual structure with IMU
[ WARN] [1501220303.977422246]: gyroscope bias initial calibration 0.000311817 -0.00545636 -0.00765611
[ INFO] [1501220303.985011783]: misalign visual structure with IMU
[ WARN] [1501220304.112688345]: gyroscope bias initial calibration -3.26388e-05   -0.0059705   0.00630322
[ INFO] [1501220304.120495876]: misalign visual structure with IMU
[ WARN] [1501220304.252251503]: gyroscope bias initial calibration  0.00129279 -0.00511213  0.00159966
[ INFO] [1501220304.260215914]: misalign visual structure with IMU
[ WARN] [1501220304.391451391]: gyroscope bias initial calibration  0.00028134 -0.00103118  -0.0122469
[ INFO] [1501220304.400189325]: misalign visual structure with IMU
[ WARN] [1501220304.514126241]: gyroscope bias initial calibration  0.00101349 -0.00391993  0.00218244
[ INFO] [1501220304.523566697]: misalign visual structure with IMU
[ WARN] [1501220304.656160601]: gyroscope bias initial calibration -3.46041e-05  -0.00299593  0.000668724
[ INFO] [1501220304.665679939]: misalign visual structure with IMU

mlyarthur, 2017-08-07:
pay attention to your input data's units
1)acc:m/s^2
2)gyo:rad/s

wangminhang, 2017-08-07:
@mlyarthur  I think it is not that problem, because it can perform very  well most time using the same hardware suit. May other reasons?

qintony, 2017-08-10:
Check you IMU data.
the units should be what @mlyarthur said.
The value is around 9.8

poaongithub, 2017-10-11:
I met same error as you did, have you solved it? @wangminhang @qintony

Torchmm, 2021-11-19:
i face the same problem ,and checked my imu data's unit as @mlyarthur , maybe some other reason cause this!

000skysky000, 2022-03-07:
> I get the same problem can u give me a hit?
thanks~~~~~~~

lgy250, 2022-07-13:
have you finish the problem? i meet the same as yours . the system initial successfully at the first but the loop detect did not work after a while

YoungHaulk, 2023-02-26:
I  met the same problem, how did you solve it?

DYK395, 2023-06-19:
Hi,I met the same problem, cause I change the integral method from median integration to Runge-kutta integration, could you please tell me how to solve this problem?

to-HaoTian, 2023-07-27:
我遇到了相同的问题，在imu内参和相机内参标定正确的情况下，运行vins-mono仍然存在轨迹乱飞的情况，你可以在你的config.yaml中将estimate_extrinsic: 的数据设置成2，即程序会自动估计外参。

estimate_extrinsic: 2   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
                        # 2  Don't know anything about extrinsic parameters. You don't need to give R,T. We will try to calibrate it. Do some rotation movement at beginning. 

如果你标定了imu和相机的外参，你可以将estimate_extrinsic: 填写为1
千万不要填写为0，除非你非常有把握

同时我还发现，启动时如果开始图形的关键点没有找好，那么后面的轨迹大概率会漂移，你可以在墙壁或者某个地方放一张标定版，启动时将相机对准标定板一段时间，然后在去拍摄其他地方，这样你的轨迹就不会漂移的太严重

Jawaheru, 2024-04-26:
I had the same issue "gyroscope bias initial calibration"and i need the solution.Note that i didnt change any thing in my data I calibrated the imu using imu_utils thin use kalibr to calibrate it with my camera. is it frames issue?