HKUST-Aerial-Robotics/VINS-Mono #63: Eigen error

hkkoo1, 2017-07-28:
thanks for yours  work!
when i use the  EuRoC dataset  On my 32 bit computer,It came with an  Eigen error，But everything is OK on my 64 bit computer, Do you know how to solve it?

there is i run log:
vins_estimator: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 270; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

qintony, 2017-08-03:
I don't know why. Eigen compatibility problem.

hkkoo1, 2017-08-03:
@qintony thanks,now i using my 64bit computer,It works  well.

heguixiang, 2017-08-14:
@hkkoo1 @qintony  ，hello，I meet the same question when running VINS on my embedded 32 bit hardware（4 cortex A17 cores），do  you solved it?

hkkoo1, 2017-08-14:
@heguixiang sorry,I changed for 64bit computer