HKUST-Aerial-Robotics/VINS-Mono #64: VINS-Mono on Nvidia TX2

maciejmatuszak, 2017-07-28:
Hi,
I am trying to run the VINS-Mono on TX2, as per comment in #1 it should work but I can not get it to converge, the position runs away. It runs perfect on laptop. I tried the MH05 sample as well as my own camera data, same deal, fine on laptop but on TX1/TX2 the position will run away. I already disabled the loop closure and limited the number of features to 100. 
I am using JetPack 3.0 with latest libraries as of today versions of all libraries are in apt-show-versions.txt.
Here is recorded session: https://youtu.be/RMSEAjrPJWM
What's the TX2 environment it was tested on?
Can I have some help/guidance how to resolve the problem

Attached are logs from the video.
[tx2_MH05.log.txt](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/files/1182141/tx2_MH05.log.txt)
[apt-show-versions.txt](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/files/1182167/apt-show-versions.txt)

maciejmatuszak, 2017-07-29:
Without knowing exactly the SW architecture I am moving by instinct.
My plan is to rebuild ceres and VINS-Mono with Eigen 3.3.3 and see if that will help.
Any other ideas?

maciejmatuszak, 2017-08-02:
Got it working on TX2 with Eigen 3.2.9 and ceres 1.11. I installed Eigen under /usr otherwise It was always selecting the system default 3.2.92. With loitor camera all 4 cores ar loaded to about 60%
Will try my TX1 next.

maciejmatuszak, 2017-08-03:
OK, TX1 runs almost ok, it is on the edge of it's abilities though. With 150 features and loop closure on there are delays in processing messages accumulating to 2-3 sec. 
With just 100 features the delays are almost gone but the solution fails when stationary:
`
                if (jacobian_pose_i.maxCoeff() > 1e8 || jacobian_pose_i.minCoeff() < -1e8)
                {
                    ROS_WARN("numerical unstable in preintegration");
                    //std::cout << sqrt_info << std::endl;
                    ROS_BREAK();
                }
`
so time to switch to TX2.
I just wonder is there a bug in Eigen past 3.2.9 or the performance of Eigen is degraded in later versions...

qintony, 2017-08-03:
Thanks for your work. We have also found that the Eigen installed by apt-get have some compatibility problem on TX2. So manually installing Eigen is a smart choice.
We have tested it with Eigen 3.3.3 on TX2. It can work. 

https://github.com/ceres-solver/ceres-solver/issues/288

maciejmatuszak, 2017-08-03:
ceres-solver/ceres-solver#288 is exactly what give me the ideas to play with Eigen. I tried the Eigen 3.3.3 and it did not work for me. After the later experiments I doubt I really used version 3.3.3 it is very easy to pick up the system default version and not notice. @qintony Is the performance experience on TX1 described above what you would expect? I run TX1 from SD card and disk access is bit slow but that should not matter at runtime.

maciejmatuszak, 2017-08-07:
One more report. I rebuild my TX2 with Jetpack 3.1 (L4T 28.1). It would diverge with default Eigen. 
I installed the 3.3.3 as system default, rebuild the ceres solver from 1.13 label and rebuild VINS-Mono. 
Works flawlessly :)!

Maciej

qintony, 2017-08-10:
It can work real time with TX1.
Sometimes you need to set less feature numbers, 100 - 150.

shishiteng, 2017-08-10:
@maciejmatuszak You should use eigen3.2 on Ubuntu16.

maciejmatuszak, 2017-08-11:
@shishiteng any particular reason?

shishiteng, 2017-08-11:
@maciejmatuszak I think eigen3.3 and higher version is not compatible with ceres in TX2  ,I replaced eigen3.3 with 3.2 then it works.

qintony, 2017-08-14:
@shishiteng Thanks for your answer!

gossterrible, 2017-10-27:
@maciejmatuszak Hi , I which to know what was your envirment setup when you where running Vins Mono on the TX1.
Here is my enviroment please let me know what i can change or what is different from yours 
- Ubuntu 16.04
- Ros Kinetic 
- opencv 2.4.8
- Eigen 3.2.92
- Ceres 1.13

Thanks

tongtybj, 2018-09-01:
@qintonguav @maciejmatuszak  

I can compile and run ```euroc``` sample in TX2 with following envrioment.

- Ubuntu 16.04
- Ros Kinetic
- opencv 3.3.1 (ros-kinetic-opencv3)
- Eigen 3.2.92
- Ceres 1.11

However, the performance is not good , there is a large drift from the beginning every trial.
Some important parameters in .yaml file are set as follows:
```
max_cnt: 120 # also tried with 150
freq: 10
loop_closure: 0
``` 

Can you give me some advise, and please tell me whether the following initialized gyroscope bias is correct for ?

wells-sde, 2024-12-02:
I encounter the same problem on a desktop pc with ubuntu 20.04, ceres-slove 1.14 and ros noetic, have you solved this?