HKUST-Aerial-Robotics/VINS-Mono #65: Big drift and ROS_WARN: wait for imu, only should happen at the beginning

marooncn, 2017-07-29:
When I use my device to run your code. It can compile and run, but the result is wired and unstable with big drift. 
[image]
(Actually, I just move my device slowly around)
My device parameter lists as below.
Camera frequency: 20HZ
IMU frequency: 400HZ (20HZ between two frames)
the offset: 3ms (as the picture shows)
[image]
I pay attention to [this issue](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/58), maybe we have the same reason and I highly expect  your  test result @qintony .
And another problem is the ROS_WARN: wait for imu, only should happen at the beginning
[image]
We have tested that the imu data can be converted to the node to deal with without lost.
So what's the problem? Thank you in advance.

maciejmatuszak, 2017-07-29:
What's the Hardware Software spec of the machine you running it on. Have you look at this issue #1?
 I am trying to run it on TX2 but no luck so far: #64
@qintony I think you will have to put something very specific on README file that the performance or ability run stable is linked to capability of the hardware.

marooncn, 2017-07-29:
@maciejmatuszak My software and  computer configuration is as mentioned in #60 . The device is customized.

guanglun, 2017-07-29:
Pay attention to two points:
The unit of acc and gyro.
acc :m/s^2;
gyro:rad/s;

marooncn, 2017-07-29:
@guanglun , Thank for your response, but the units have been transformed to be metric.

marooncn, 2017-08-08:
Solution:
The problem is finally solved. There is three points.
1. Pay attention to the time. The IMU should be pre-integrated， which means IMU timestamp is more faster than camera  and the last timestamp is  the same as the latest camera's in each frame.
2. In thesis, the gyro bias is assumed to zero mean. So you'd better calculate the actual mean and minus it. you can use [https://github.com/marooncn/imu_filter/blob/master/src/bias_calculator.cpp](url) to calculate the bias mean.
3. In config file, the  projection_parameters means camera intrinsic matrix parameters.

qintony, 2017-08-10:
Thanks for your work.
Some points which I have different ideas.

2. The gyro bias is not assumed to zero mean. We calibrate is online. The gyro scope noise is assumed to be zero mean. You can check the gyr bias estimation result on EuRoC result.

4. You can test "set estimate_extrinsic to 1 or 2" with EuRoC dataset. In fact, I always use 1 on my own sensors. Sometimes I use 2, since I am lazy to do extrinsic parameter offline calibration.

marooncn, 2017-08-11:
@qintony Thanks for your response.
 2.  maybe my expression is not right. My original thought is the gyro mean when it's static.
 4.  I set estimate_extrinsic to  2 with EuRoC dataset to run the demo and find it's truly right. But for my dataset, it's invalid just as the below pictures show.
[image]
estimate_extrinsic : 0
[image]
estimate_extrinsic : 1
[image]
estimate_extrinsic : 2
One key difference between my dataset and EuRoC is  the IMU. My device's IMU is ICM20602($5.72) and they use ADIS16448($670.82). Maybe their device is more reliable and the optimization algorithm is based on much reliable data. And I revise the 4. 
Another question is the scale. We find the scale is much smaller than the actual when we run the code outdoors just as picture 1 shows.

qintony, 2017-08-11:
Yes. The second question shows that your IMU is unsynchronized or in low quality. That also explains that 4 fails with your sensor set.
In fact, I find out even the most cheapest IMU (mpu6050), if well synchronized (https://www.seeedstudio.com/Loitor-Cam2pc-Visual-Inertial-SLAM-p-2873.html), can run VINS well.

marooncn, 2017-08-11:
Thank you !
And what about the scale, do you ever test the code outdoors with big scale, is it OK in the scale?

marooncn, 2017-08-11:
@qintony I test with Loitor  in the afternoon and find it truly works well even though I set estimate_extrinsic to 1 or 2 except the scale is a little smaller than the actual (when estimate_extrinsic = 0, the scale is right). I am so weird that ICM20602 is more expensive than MPU6050, but why does the MPU6050 works much better? And our device is according to [the device getting EuRoC dataset ](https://www.researchgate.net/publication/265683167_A_Synchronized_Visual-Inertial_Sensor_System_with_FPGA_Pre-Processing_for_Accurate_Real-Time_SLAM) to make sure the synchronization. 
I pay attention to that  MPU6050 uses DMP to deal with the original data, is something with it? Because we don't deal with the original data from IMU . And if we use filter to deal with it, does it effect the synchronization?

qintony, 2017-08-14:
Sorry, I know little about hardware level. I don't know how IMU works...  Sorry about that.

What I know is that I fully test the time offset between IMU and camera in Loitor. The time offset is stable and small, 3-5 ms.

The test method is that the camera look at chess board, meanwhile I rotate the device.
We can get two ratation, one is from vision (chess board), one is from IMU propagation.
We can find a time offset which can minimize these two rotations.
You can test your device by this way.

guanglun, 2017-08-20:
@marooncn @qintony 
    The IMU Ros node is include acc_xyz,gyro_xyz.That's true.But are the orientation is useful?I found The EuRoC dataset orientaion is 1 0 0 0 always.
    About the synchronized.Because I use the hardware of mine:MT09V034 and MPU6000(MPU6000 is same as MPU6050).The hardware is same as loitor.I use an 1ms timer to add up.Each IMU or image data send with an time-stamp.The synchronized is good I think.But I can't work well.
    Next is the log of time-stamp:

`
Image:1755 340000000
IMU:1755 366000000
IMU:1755 371000000
IMU:1755 375000000
IMU:1755 380000000
IMU:1755 384000000
IMU:1755 389000000
IMU:1755 393000000
Image:1755 372000000
IMU:1755 398000000
IMU:1755 402000000
IMU:1755 411000000
IMU:1755 416000000
IMU:1755 420000000
IMU:1755 425000000
IMU:1755 429000000
Image:1755 404000000
IMU:1755 433000000
IMU:1755 438000000
IMU:1755 443000000
IMU:1755 447000000
IMU:1755 456000000
IMU:1755 460000000
IMU:1755 465000000
Image:1755 436000000
IMU:1755 470000000
IMU:1755 474000000
IMU:1755 479000000
IMU:1755 483000000
IMU:1755 487000000
IMU:1755 492000000
IMU:1755 501000000
Image:1755 500000000
IMU:1755 506000000
IMU:1755 510000000
IMU:1755 514000000
IMU:1755 519000000
IMU:1755 523000000
IMU:1755 528000000
IMU:1755 533000000
Image:1755 532000000
IMU:1755 537000000
IMU:1755 541000000
IMU:1755 550000000
IMU:1755 555000000
IMU:1755 559000000
IMU:1755 564000000
IMU:1755 568000000
Image:1755 564000000
IMU:1755 573000000
IMU:1755 577000000
IMU:1755 582000000
IMU:1755 591000000
IMU:1755 595000000
IMU:1755 600000000
IMU:1755 604000000
Image:1755 596000000
IMU:1755 609000000
IMU:1755 613000000
IMU:1755 618000000
IMU:1755 622000000
IMU:1755 627000000
IMU:1755 631000000
IMU:1755 640000000
Image:1755 628000000
IMU:1755 645000000
IMU:1755 649000000
IMU:1755 654000000
IMU:1755 658000000
IMU:1755 663000000
IMU:1755 667000000
IMU:1755 672000000
Image:1755 660000000
IMU:1755 676000000
IMU:1755 685000000
IMU:1755 690000000
IMU:1755 694000000
IMU:1755 699000000
IMU:1755 703000000
IMU:1755 708000000
Image:1755 692000000
IMU:1755 712000000
IMU:1755 717000000
^CIMU:1755 721000000
IMU:1755 730000000
IMU:1755 735000000
IMU:1755 739000000
IMU:1755 744000000
Image:1755 724000000
IMU:1755 748000000
IMU:1755 753000000
IMU:1755 757000000
IMU:1755 762000000
IMU:1755 766000000
IMU:1755 775000000
IMU:1755 780000000
Image:1755 756000000
IMU:1755 784000000
IMU:1755 789000000
IMU:1755 793000000
IMU:1755 798000000
IMU:1755 802000000
IMU:1755 807000000
IMU:1755 811000000
Image:1755 788000000
IMU:1755 816000000
IMU:1755 825000000
IMU:1755 829000000
IMU:1755 834000000
IMU:1755 838000000
IMU:1755 843000000
IMU:1755 847000000
Image:1755 820000000
IMU:1755 852000000
IMU:1755 856000000
IMU:1755 861000000
IMU:1755 870000000
IMU:1755 874000000
IMU:1755 879000000
IMU:1755 883000000
Image:1755 852000000
IMU:1755 888000000
IMU:1755 892000000
IMU:1755 897000000
IMU:1755 901000000
IMU:1755 906000000
IMU:1755 915000000
IMU:1755 919000000
Image:1755 916000000
IMU:1755 924000000
IMU:1755 928000000
IMU:1755 933000000
IMU:1755 937000000
IMU:1755 942000000
IMU:1755 946000000
`
1755 is sec,946000000 is nsc.
    I can't found what's problem with it.I will be happy to hear your advices.

qintony, 2017-08-26:
Orientation is not useful. Orientation is the result fused by IMU itself. We don't  what fusion method is used and we don't how accurate it is.

qintony, 2017-08-26:
Timestamp looks ok. But is the timestamp accurate? May transmission delay, trigger delay influence this timestamp?