HKUST-Aerial-Robotics/VINS-Mono #66: How to calibrate a fisheye camera and write the parameter in the right format?

kevinchow1993, 2017-08-05:
hi, thanks for your work,
I tried  to use a **fisheye camera**, I use your camera calibration node to calibrate it, and  i got the result like this
```
model_type: MEI
camera_name: mv_25002267
image_width: 752
image_height: 480
mirror_parameters:
   xi: 1.3672007173379490e+00
distortion_parameters:
   k1: 2.7080458705446409e-01
   k2: -8.7747378654920036e-02
   p1: 1.7312161496315188e-03
   p2: -6.9293546227690240e-04
projection_parameters:
   gamma1: 4.1922203248509197e+02
   gamma2: 4.1922889258400278e+02
   u0: 3.7760726402265072e+02
   v0: 2.5436221559052447e+02
```
so , i write the parameter in the yaml file in  this format,
is that right?

kevinchow1993, 2017-08-05:
Here is my yaml file 
```
%YAML:1.0

#common parameters
imu_topic: "/mavros/imu/data"
image_topic: "/mv_25002267/image_raw"
output_path: "/config/euroc/vins_result.csv"         # vins outputs will be wrttento vins_folder_path + output_path

#camera calibration 
model_type: MEI
camera_name: mv_25002267
image_width: 752
image_height: 480
mirror_parameters:
   xi: 1.3672007173379490e+00
distortion_parameters:
   k1: 2.7080458705446409e-01
   k2: -8.7747378654920036e-02
   p1: 1.7312161496315188e-03
   p2: -6.9293546227690240e-04
projection_parameters:
   gamma1: 4.1922203248509197e+02
   gamma2: 4.1922889258400278e+02
   u0: 3.7760726402265072e+02
   v0: 2.5436221559052447e+02

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
equalize: 1             # if image is too dark or light, trun on equalize to find enough features
fisheye: 1              # if using fisheye, trun on it. A circle mask will be loaded to remove edge noisy points

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.2          # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.02         # gyroscope measurement noise standard deviation.     #0.05
acc_w: 0.0002         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.81007     # gravity magnitude

#loop closure parameters
loop_closure: 1   #if you want to use loop closure to minimize the drift, set loop_closure true and give your brief pattern file path and vocabulary file path accordingly;
                     #also give the camera calibration file same as feature_tracker node
pattern_file: "/support_files/brief_pattern.yml"
voc_file: "/support_files/brief_k10L6.bin"
min_loop_num: 25
```

wangminhang, 2017-09-13:
@kevinchow1993 how can you compile Calibration or CamOdoCal？
 when i try to compile it，i have a error “recipe for target 'all' failed” ， how to debug？
thak you

shengxingdong, 2017-10-13:
@kevinchow1993  is your method right?