HKUST-Aerial-Robotics/VINS-Mono #68: Maybe a bug. visualInitialAlign() frame_i->second.R * x.segment<3>(kv * 3) out of range

Zekegao, 2017-08-07:
Hi,
I found in the visualInitialAlign(), maybe there is a bug that Vs[kv] = frame_i->second.R * x.segment<3>(kv * 3);
Run it, kv could be 4, but x.segment<3> just 12 elements, so it will out of range.
But, I cannot confirm it. Because I run it without ROS, so maybe my modification led to this bug.
My log as below
[image]

Zekegao, 2017-08-07:
So, I add a condition, now I can run it. But, I don t  know if this condition could affect all system.
```
if (kv == x.size()/3)
                break;
Vs[kv] = frame_i->second.R * x.segment<3>(kv * 3);
```

qintony, 2017-08-10:
I have checked it on my code. 
x.size always large than kv * 3.
The extrem case is that x size is 36
kv 0 - 10
x.size = 11 * 3 (velocity) + 2 (gravity) + 1 (scale)
[image]

Anyway, add your condition to keep safe.

Zekegao, 2017-08-10:
@qintony Thx for your reply.
I also found the Camera Matrix K in lots of place are not be set. it's usually 1 0 0 , 0 1 0 , 0 0 1.
Should I change it to myself camera parameters?
Actually, when I changed, the Ceres Solver will not convergence, and the final_cost also bigger than 5e-03, so the the GlobalSFM::construct will return false.
If I need not to change the the camera parameters, why you didn't use the parameters by calibrate to calculate the cv::recoverPose and solvePnPRansac? Is there some reason?
Best,
zeke

qintony, 2017-08-10:
You don't need to do that change.
Because in the feature_tracker node, the features are already undistorted by K and D, which is normalized points.

If you think  5e-03 is too strict, you can release it.

Zekegao, 2017-08-10:
@qintony Got it, Thank you very much!
Another question....
After complete Initialization, the jacobian_speedbias_i.maxCoeff() will very very big, and then display Assertion `fabs(jacobian_speedbias_i.maxCoeff()) < 1e8' failed.
I have checked, so I found the value of all_image_frame.second.R is big, like  
2000.58 -973.176  -9.6511
 229.516   -2.339 -5.24738
-15.8506  9.85859        0
I dont know why, I check the calib_ric is seem like normal. relative_R  in relativePose is also correct.
Anyway, thank you very much.
Best,
zeke