HKUST-Aerial-Robotics/VINS-Mono #70: Error "[vins_estimator-2] process has died " when running MH_05_difficult dataset sample.

LittleMing233, 2017-08-07:
Hi ~
I have run the MH_05_difficult dataset on Workstation Ubuntu 16.04 ,and it can tracking well and make a good trajectory.
But when I copy the packge(the same code from workstation) to the onboard PC  , it always have an error.The terminal Shows:
  
[image]

But the tracking thread is working well, it can tracking in real time. 
[image]
My Onboard PC is Intel NUC with i7 CPU.
Software : Ubuntu 16.04 and ROS Kinetic version.(The same as my Workstation) 

Can you solve it ?
Thx~

qintony, 2017-08-10:
Take care of the Eigen and Ceres version.

Make it same with you workstation.

LittleMing233, 2017-08-12:
OK，I have installed the eigen3.2.10 and rebuild Ceres ,it works well on the NUC now.
Thank you!

closed 2017-08-12 by LittleMing233