HKUST-Aerial-Robotics/VINS-Mono #71: solveGyroscopeBias

shopkauf, 2017-08-07:
Hi,
I am trying to understand the math in the function solveGyroscopeBias( ), it seems different from the ISMAR'17 paper (Sec 6.2.1 Gyroscope Bias), and also different from the VINS-Mono technical report Eq. 14. 
Your help is very much appreciated, if you could point me to the correct equations. 
Thanks!

qintony, 2017-08-10:
It's totally same with Eq 14 in the technical report.

[image]

[image]

tmp_A is jacobian

tmp_b is 
[image]

luodanping, 2019-05-29:
Hi,
It is hard for me to understand this assignment expression.
Attempts tried by searching the internet,but still failed..
1. block is a member function of Matrix in Eigen to get a block of matrix, but what is it here?
2. what is .template? and what is the syntax between .template and block?
tmp_A = frame_j->second.pre_integration->jacobian.template block<3, 3>(O_R, O_BG);

Can anybody tell me..
Thanks

jjc7151, 2019-10-11:
I think, frame_i->second.pre_integration->delta_q is omitted 
in tmp_b = 2 * (frame_j->second.pre_integration->delta_q.inverse() * q_ij).vec();

tmp_b = 2 * (frame_j->second.pre_integration->delta_q.inverse() * frame_i->second.pre_integration->delta_q * q_ij).vec();

chennuo0125-HIT, 2020-05-29:
@luodanping @shopkauf you can see this blog [https://blog.csdn.net/qq_41839222/article/details/89106128](url), explain perfect!

cheslee-z, 2023-10-02:
> Hi, It is hard for me to understand this assignment expression. Attempts tried by searching the internet,but still failed..
> 
> tmp_A = frame_j->second.pre_integration->jacobian.template block<3, 3>(O_R, O_BG);
> 
> 1. block is a member function of Matrix in Eigen to get a block of matrix, but what is it here?
> 2. what is .template? and what is the syntax between .template and block?
> 
> Can anybody tell me.. Thanks

I also encounter the same problem, have you solved it?

luodanping, 2023-10-02:
您好，      您的邮件我已收到，我将尽快回复您。

cheslee-z, 2023-10-02:
> I also encounter the same problem, have you solved it?

maybe you could refer this instruction https://eigen.tuxfamily.org/dox/TopicTemplateKeyword.html

luodanping, 2023-10-08:
> maybe you could refer this instruction https://eigen.tuxfamily.org/dox/TopicTemplateKeyword.html

Thanks for your share.

luodanping, 2023-10-08:
> maybe you could refer this instruction https://eigen.tuxfamily.org/dox/TopicTemplateKeyword.html

Thanks for your kindly share. It's the right reference for someone who runs into the same problem