HKUST-Aerial-Robotics/VINS-Mono #72: Might be a logic error in Estimator.cpp

Doodleshr, 2017-08-08:
In the last few lines of Estimator.cpp 

`std::unordered_map<long, double *> addr_shift;
            for (int i = 0; i <= WINDOW_SIZE; i++)
            {
                if (i == WINDOW_SIZE - 1)
                    continue;
                else if (i == WINDOW_SIZE)
                {
                    addr_shift[reinterpret_cast<long>(para_Pose[i])] = para_Pose[i - 1];
                    addr_shift[reinterpret_cast<long>(para_SpeedBias[i])] = para_SpeedBias[i - 1];
                }
                else
                {
                    addr_shift[reinterpret_cast<long>(para_Pose[i])] = para_Pose[i];
                    addr_shift[reinterpret_cast<long>(para_SpeedBias[i])] = para_SpeedBias[i];
                }
            }`

Firstly I noticed this loop runs 11 times starting from i=0 to i=10 .
When i==9, the loop continues and para_Pose[9] (the 10th element) is not assigned with any value; when i==10,  para_Pose[9] (the 10th element) is assigned to para_Pose[10](the 11th element) .
I think it might be a error since the size of para_Pose should be 10.

S

qintony, 2017-08-10:
It's the marginalization framework.
We solve 10 frames and 1 newest frame, totally 11 frames in the optimization.
If the second newest frame is non-keyframe, we will throw it.

Doodleshr, 2017-08-10:
@qintony Thanks, it makes sense.

closed 2017-08-10 by Doodleshr