HKUST-Aerial-Robotics/VINS-Mono #73: Question about the code in initial_sfm.cpp

yyIsMe, 2017-08-08:
Hi，
   Great job you guys did, I tried to understand following code，and i got stuck in the function--bool GlobalSFM::construct().
Why did you use solveFrameByPnP() and triangulateTwoFrames() for many times? Is that any purpose  you wanna doing? Could you explain it?
The code as follow:
`//1: trangulate between l ----- frame_num - 1
	//2: solve pnp l + 1; trangulate l + 1 ------- frame_num - 1; 
	for (int i = l; i < frame_num - 1 ; i++)
	{
		// solve pnp
		if (i > l)
		{
			Matrix3d R_initial = c_Rotation[i - 1];
			Vector3d P_initial = c_Translation[i - 1];
			if(!solveFrameByPnP(R_initial, P_initial, i, sfm_f))
				return false;
			c_Rotation[i] = R_initial;
			c_Translation[i] = P_initial;
			c_Quat[i] = c_Rotation[i];
			Pose[i].block<3, 3>(0, 0) = c_Rotation[i];
			Pose[i].block<3, 1>(0, 3) = c_Translation[i];
		}

		// triangulate point based on the solve pnp result
		triangulateTwoFrames(i, Pose[i], frame_num - 1, Pose[frame_num - 1], sfm_f);
	}
	//3: triangulate l-----l+1 l+2 ... frame_num -2
	for (int i = l + 1; i < frame_num - 1; i++)
		triangulateTwoFrames(l, Pose[l], i, Pose[i], sfm_f);
	//4: solve pnp l-1; triangulate l-1 ----- l
	//             l-2              l-2 ----- l
	for (int i = l - 1; i >= 0; i--)
	{
		//solve pnp
		Matrix3d R_initial = c_Rotation[i + 1];
		Vector3d P_initial = c_Translation[i + 1];
		if(!solveFrameByPnP(R_initial, P_initial, i, sfm_f))
			return false;
		c_Rotation[i] = R_initial;
		c_Translation[i] = P_initial;
		c_Quat[i] = c_Rotation[i];
		Pose[i].block<3, 3>(0, 0) = c_Rotation[i];
		Pose[i].block<3, 1>(0, 3) = c_Translation[i];
		//triangulate
		triangulateTwoFrames(i, Pose[i], l, Pose[l], sfm_f);
	}
	//5: triangulate all other points
	for (int j = 0; j < feature_num; j++)
	{
		if (sfm_f[j].state == true)
			continue;
		if ((int)sfm_f[j].observation.size() >= 2)
		{
			Vector2d point0, point1;
			int frame_0 = sfm_f[j].observation[0].first;
			point0 = sfm_f[j].observation[0].second;
			int frame_1 = sfm_f[j].observation.back().first;
			point1 = sfm_f[j].observation.back().second;
			Vector3d point_3d;
			triangulatePoint(Pose[frame_0], Pose[frame_1], point0, point1, point_3d);
			sfm_f[j].state = true;
			sfm_f[j].position[0] = point_3d(0);
			sfm_f[j].position[1] = point_3d(1);
			sfm_f[j].position[2] = point_3d(2);
			//cout << "trangulated : " << frame_0 << " " << frame_1 << "  3d point : "  << j << "  " << point_3d.transpose() << endl;
		}		
	}`

qintony, 2017-08-10:
For bundle adjustment, the key is to make the initial guess as good as possible.
We do pnp and triangulation many times is to get the initial guess of all poses and features.

yyIsMe, 2017-08-11:
Thanks for your answer, qin. And I have another question in paper：
[image]
Why did you directly using the Euler Angle without converting it to quaternion?

qintony, 2017-08-14:
One issue of Euler angle is Gimbal lock...
Another issue is multiple combinations of Euler angle correspond to one rotation...
In usual, we use quaternion or SO(3).

yyIsMe, 2017-08-22:
Thanks,@qintony, now I get a new trouble. Can you express the following code? It seems not  corresponding to the article.
[image]
[image]

qintony, 2017-08-26:
[image]

same meaning.

yyIsMe, 2017-08-28:
Thank you very much,@qintony, I got it!