HKUST-Aerial-Robotics/VINS-Mono #74: How to obtain covariance of the pose

zjcv2012, 2017-08-08:
Hi,
I want to obtain the covariance of the odometry data(position+orientation). I tried to use the covariance estimation in ceres solver, the code is like:
[image]

The problem is that if I use SPARSE_QR(default) as the algorithm type, the error called Jacobian matrix is rank deficient, and if I use DENSE_SVD(which can deal with deficient rank problem), it takes too much time. Does any of you have some good suggestions?

Thanks.

qintony, 2017-08-10:
Thanks for your question. We don't have a good way to get covariance yet.
I will let you know if we figure it out in the future.

zjcv2012, 2017-08-10:
@qintony Thanks, I will appreciate that.

wangminhang, 2017-09-08:
@zjcv2012 I also want to calculate the pose covariance, could you show more code about your implement，thanks

zjcv2012, 2017-09-08:
@wangminhang Hi, I don't actually solve it. But I saw some posts from the Internet that the latest ceres library(not the stable version) has the function to get the covariance for this case.

zhanghanduo, 2019-04-23:
@zjcv2012 I changed the linear_solver_type option from DENSE_SCHUR to DENSE_QR, it turned out that the deficiency issue is no longer there. But the processing time for sparse qr covariance estimation is still not fast (about 20 ms) which generates a lot of delays. 

I don't recommend using this method to calculate covariance values, but I still cannot find a better way. In my opinion, covariance value should be easily calculated with some intrinsic values like computeMarginals in g2o without too much efforts.

maltoseslope, 2019-08-29:
> @zjcv2012 I changed the linear_solver_type option from DENSE_SCHUR to DENSE_QR or SPARSE_SCHUR, it turned out that the deficiency issue is no longer there. But the processing time for sparse qr covariance estimation is still not fast (about 20 ms) which generates some delays. As Dense_QR is not suitable for BA, you can choose SPARSE_SCHUR as alternative.

Hi, thanks for your answer! Have you tried on VINS FUSION to get the covariance? Do you mind post your code to get the covariance?

maltoseslope, 2019-08-29:
@zhanghanduo Hi, thanks for your answer! Have you tried on VINS FUSION to get the covariance? Do you mind post your code to get the covariance?

zhanghanduo, 2019-08-29:
I just replaced SPARSE_QR with DENSE_SCHUR (use newer version of ceres), commented `trust_region_strategy` . The reason and how to optimize this process has not been considered yet because I am recently very busy. It's just for your reference. 

`
ceres::Solver::Options options;

//  options.linear_solver_type = ceres::DENSE_SCHUR;
options.linear_solver_type = ceres::ITERATIVE_SCHUR;
options.preconditioner_type = ceres::SCHUR_JACOBI;
options.use_explicit_schur_complement = true;
options.num_threads = 6;
//  options.trust_region_strategy_type = ceres::DOGLEG;
options.max_num_iterations = NUM_ITERATIONS;
//  options.minimizer_progress_to_stdout = true;

if (marginalization_flag == MARGIN_OLD)
    options.max_solver_time_in_seconds = SOLVER_TIME * 0.8;
else
    options.max_solver_time_in_seconds = SOLVER_TIME;

TicToc t_solver;
ceres::Solver::Summary summary;
ceres::Solve(options, &problem, &summary);

// Covariance Estimation!
TicToc t_cov;

cout << "Total cost time: " << summary.total_time_in_seconds <<
" | linear solver: " << summary.linear_solver_time_in_seconds <<
" | TrustRegionMinimizer time: " << summary.minimizer_time_in_seconds << endl;
printf("solver costs: %f \n", t_solver.toc());

// Covariance of poses
ceres::Covariance::Options cov_options;
cov_options.num_threads = 6;
ceres::Covariance covariance(cov_options);

std::vector<std::pair<const double *, const double *>> covariance_blocks;
covariance_blocks.emplace_back(para_Pose[WINDOW_SIZE], para_Pose[WINDOW_SIZE]);
//        CHECK(covariance.Compute(covariance_blocks, &problem));
if(covariance.Compute(covariance_blocks, &problem)) {
    double covariance_pose[SIZE_POSE * SIZE_POSE];
    covariance.GetCovarianceBlock(para_Pose[WINDOW_SIZE], para_Pose[WINDOW_SIZE], covariance_pose);

    for (auto x = std::begin(covariance_pose); x != std::end(covariance_pose);)
        cout << *++x << " " << endl;
    printf("covariance solver costs: %f \n", t_cov.toc());
}
`

maltoseslope, 2019-08-30:
@zhanghanduo Thanks a lot for your reply! It works in VINSMono, but in VINSFusion it reports error: "Check failed: it != collection.end() Map key not found: 0x87f668" at the line "covariance.Compute(covariance_blocks, &problem)", by any chance you know the reason?

DanielCyberbee, 2024-11-25:
@maltoseslope have you figured the reason why it happens?