HKUST-Aerial-Robotics/VINS-Mono #75: can't roslaunch vins_estimator euroc.launch

csufangyu, 2017-08-10:
Hi
thanks for your job.Your job is very great and I want to try to run your code.but I  run the program meet error when I try to run "roslaunch vins_estimator euroc.launch".The error is:
fangyu@fangyu-ThinkPad-Edge-E431:~$ roslaunch vins_estimator euroc.launch
... logging to /home/fangyu/.ros/log/940d0962-7db4-11e7-9556-28d244755860/roslaunch-fangyu-ThinkPad-Edge-E431-8913.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fangyu-ThinkPad-Edge-E431:41347/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/fangyu/catk...
 * /feature_tracker/vins_folder: /home/fangyu/catk...
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /vins_estimator/config_file: /home/fangyu/catk...
 * /vins_estimator/vins_folder: /home/fangyu/catk...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    vins_estimator (vins_estimator/vins_estimator)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[feature_tracker-1]: started with pid [8931]
process[vins_estimator-2]: started with pid [8932]
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

[ INFO] [1502360204.590471499]: init begins
[ INFO] [1502360204.596630050]: Loaded config_file: /home/fangyu/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
[ INFO] [1502360204.597450319]: Loaded vins_folder: /home/fangyu/catkin_ws/src/VINS-Mono/feature_tracker/../config/../
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

[feature_tracker-1] process has died [pid 8931, exit code -6, cmd /home/fangyu/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/fangyu/.ros/log/940d0962-7db4-11e7-9556-28d244755860/feature_tracker-1.log].
log file: /home/fangyu/.ros/log/940d0962-7db4-11e7-9556-28d244755860/feature_tracker-1*.log
[vins_estimator-2] process has died [pid 8932, exit code -6, cmd /home/fangyu/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/fangyu/.ros/log/940d0962-7db4-11e7-9556-28d244755860/vins_estimator-2.log].
log file: /home/fangyu/.ros/log/940d0962-7db4-11e7-9556-28d244755860/vins_estimator-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
How can I solve this problem?
very thanks!!!

qintony, 2017-08-14:
It same to be some Opencv issue.

Check whether this path "/home/fangyu/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml" exist.

Try the code on another computer. Make sure 64 bit system.

csufangyu, 2017-08-14:
Thanks for your help.There is OpenCv3.1 on my computer,and these files all exist.Maybe opencv issue.

gossterrible, 2017-10-18:
Hi Did you fix your issue ?I'l having the same issue

gossterrible, 2017-10-18:
@csufangyu I had the same issue and you can refer to this [issue](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/107) for a fix. Indeed it's a compatibility issue with your opencv version

csufangyu, 2017-10-19:
thanks very much!

Heyrenqiang, 2018-11-26:
my opencv version 4.0 ，the same problem...

linuxscofield, 2020-07-04:
In my case(opencv version is 3.3.3), the reason is that I use `catkin_make_isolated` instead of `catkin_make` to compile project. After I compile with `catkin_make`, no error occurs.