HKUST-Aerial-Robotics/VINS-Mono #76: 设备问题

stevechan1993, 2017-08-12:
请问哪里可以获取到config文件夹中提到的相关设备？

LittleMing233, 2017-08-12:
please search the device on Taobao.com

stevechan1993, 2017-08-13:
淘宝、亚马逊都试过了，没找到

qintony, 2017-08-14:
You can search Loitor on Taobao, I will add one Loitor config file and launch file later.

guanglun, 2017-08-19:
@qintony 
What time would Loitor config file and launch file provide?
I have bought one.Hope to got good performance as yours.

qintony, 2017-08-26:
I have upload. @guanglun

takumijash, 2017-12-13:
@qintonguav 
Do you know how do Loitor synchronize the camera and IMU? I read the LoitorUserManual Hardware Architecture but I can't figure out how.  
One more question, what image format is it? YUV or compressed image?

Thank you.

yukti-nxg, 2018-01-31:
@qintonguav 
I am also testing on Loitor. I am unable to find your Loitor config and launch files. Currently the code fails in visual inertial alignment and scale estimation. Can you please share your Loitor config and launch files.
Thank you.